NVIDIAGameWorks/PhysX #457: Caculate center of mass in PxMassProperties.h conflicts with that in ExtRigidBodyExt.cpp

bingbingzhenbang, 2021-08-03:
in PxMassProperties.h line 138
`centerOfMass = s.rotation.rotate(s.scale.multiply(s.rotation.rotateInv(unscaledCoM)));`

while in ExtRigidBodyExt.cpp line 216
`convCoM = g.scale.rotation.rotateInv(g.scale.scale.multiply(g.scale.rotation.rotate(convCoM)));`

either of these two ways must be wrong

preist-nvidia, 2021-08-05:
Hi Rubing,

Thank you for spotting and reporting this bug, we will need to fix this.

The ExtRigidBodyExt.cpp code is correct. The PxMassProperties code is incorrect. If you fix line 138 in PxMassProperties to be

    centerOfMass = s.rotation.rotateInv(s.scale.multiply(s.rotation.rotate(unscaledCoM)));

it will be correct and consistent with the rest of the codebase.

The `PxMeshScale::toMat33` and `PxMeshScale::transform` are consistent. The API doc is not perfectly clear on this, but the PxMeshScale rotation quaternion must represent the rotation from the scaling frame to the local mesh frame, so transforming a mesh-frame vert to the scaling frame, if the quaternion's rotate method is applied to the mesh-frame vert. The COM is just like another mesh-vert that needs to be transformed.

The [transform method of PxMeshScale](https://github.com/NVIDIAGameWorks/PhysX/blob/4.1/physx/include/geometry/PxMeshScale.h#L150) does:

`rotation.rotateInv(scale.multiply(rotation.rotate(v)));`

which rotates the mesh-local vertex v to the scaling frame, scales it element-wise, and then rotates the scaled vertex back to the mesh frame.

The [toMat33 method](https://github.com/NVIDIAGameWorks/PhysX/blob/4.1/physx/include/geometry/PxMeshScale.h#L130) results in the same sequence:
    
    PxMat33 rot(rotation);
    PxMat33 trans = rot.getTranspose();
    trans.column0 *= scale[0];
    trans.column1 *= scale[1];
    trans.column2 *= scale[2];
    return trans * rot;

The operation toMat33() * v expands to trans * rot * v, with trans = rotInv * scale with scale a diagonal scaling matrix, so the same operation.

bingbingzhenbang, 2021-08-06:
> Hi Rubing,
> 
> Thank you for spotting and reporting this bug, we will need to fix this.
> 
> The ExtRigidBodyExt.cpp code is correct. The PxMassProperties code is incorrect. If you fix line 138 in PxMassProperties to be
> 
> ```
> centerOfMass = s.rotation.rotateInv(s.scale.multiply(s.rotation.rotate(unscaledCoM)));
> ```
> 
> it will be correct and consistent with the rest of the codebase.
> 
> The `PxMeshScale::toMat33` and `PxMeshScale::transform` are consistent. The API doc is not perfectly clear on this, but the PxMeshScale rotation quaternion must represent the rotation from the scaling frame to the local mesh frame, so transforming a mesh-frame vert to the scaling frame, if the quaternion's rotate method is applied to the mesh-frame vert. The COM is just like another mesh-vert that needs to be transformed.
> 
> The [transform method of PxMeshScale](https://github.com/NVIDIAGameWorks/PhysX/blob/4.1/physx/include/geometry/PxMeshScale.h#L150) does:
> 
> `rotation.rotateInv(scale.multiply(rotation.rotate(v)));`
> 
> which rotates the mesh-local vertex v to the scaling frame, scales it element-wise, and then rotates the scaled vertex back to the mesh frame.
> 
> The [toMat33 method](https://github.com/NVIDIAGameWorks/PhysX/blob/4.1/physx/include/geometry/PxMeshScale.h#L130) results in the same sequence:
> 
> ```
> PxMat33 rot(rotation);
> PxMat33 trans = rot.getTranspose();
> trans.column0 *= scale[0];
> trans.column1 *= scale[1];
> trans.column2 *= scale[2];
> return trans * rot;
> ```
> 
> The operation toMat33() * v expands to trans * rot * v, with trans = rotInv * scale with scale a diagonal scaling matrix, so the same operation.
> 
> Best,
> Philipp

Thanks for you explanation.

preist-nvidia, 2021-08-06:
Thank you again for reporting - I fixed the issue internally for future releases.

closed 2021-08-06 by preist-nvidia