NVIDIAGameWorks/PhysX #459: PhysX SDK dead?

Zeblote, 2021-08-07:
https://developer.nvidia.com/physx-sdk

Page now reads: `PhysX 5 is exclusively available to the public in the NVIDIA Omniverse platform.`

Zeblote, 2021-08-09:
I really don't understand it. What is supposed to be the point of it being exclusive to Omniverse? Being able to set up my simulation in Omniverse is entirely useless as I can't then run the same simulation in Unreal Engine or anything else. Or are you trying to say everyone should migrate from Unreal, Unity, etc to using Omniverse as their game engine and ship products made with only that? Is that even possible?

This is especially confusing as you previously said on here that an SDK would become available.

Esentiel, 2021-08-10:
I spent some time today trying to understand what Omniverse is and what it's not.
I assume, it is advanced scene editor with real time collaboration, physics simulation, cloud storage, facial animation etc.
But it's not game engine, it's more like asset creation tool.
Say I wanna utilize it, so I made some model with physics, joints, cloth. How can I use it in my game if my game engine doesn't have PhysX 5?

Zeblote, 2021-08-10:
You can't! Amazing, right?

Esentiel, 2021-08-10:
If that's the case, that seems kind of counter logical to me, as a game developer.
Probably omniverse wasn't meant for gamedev, probably cartoons, engineering etc., where you don't need simulation outside scene editor idk.

Zeblote, 2021-08-10:
> there is typo on the linked page :(
The page has many errors, such as calling it an SDK, when none exists.

Zeblote, 2021-08-11:
There are only two things I need from PhysX 5:
- the custom collision callback thing.
- all of the misc optimizations that were made since PhysX 4.1.

All this soft body stuff, Omniverse, etc., I have no use for at all. That's not to say those things aren't really cool! But now their existence seems to be obstructing people from ever getting access to the rest. I don't understand why.

Zeblote, 2021-08-11:
The most infuriating thing about this is the complete lack of communication. First you say PhysX 5 is just around the corner - two years ago. Then you say an SDK will be available soon. Then you said PhysX is exclusive to Omniverse for some inexplicable reason. Which is it? Is there any point in waiting for it?

samvv, 2021-08-20:
@Zeblote Totally agree with you. Though I doubt NVIDIA cares about these issues.

Zeblote, 2021-08-23:
[image]

StWooow, 2021-08-25:
This is why I strongly oppose NVIDIA's acquisition of Arm. The strategy of changing at any time makes me very worried that the mobile phone with Arm will be as expensive and difficult to buy as the graphics card

SephirothFFKH, 2021-09-01:
What even more strange in this SDK page is that the "Industrial Applications for Features" section has a column for "Game Development" and all PhysX 5 features have a check mark underneath.

I would be happy with PhysX 5 being Omniverse exclusive if there was the ability to use it as a game engine with support for all GPU vendors. I mean compared to UE5, PhysX 5 is not the only improvement in Ominiverse, there is also RTX mGPU that I have been trying to do with UE5's Raytracing DX12 mGPU but it just doesn't work. Meanwhile doing it in Omniverse has been so smooth and functional.