NVIDIAGameWorks/PhysX #461: Drive Rotation limits should adjust with local frame Rotation in D6 joint.

AdilKhurshid, 2021-08-13:
Hi ! I'm trying to create a D6Joint with only TWIST motion along x Axis and rotating local frame Transform with some rotation while creating joint. After joint creation I'm setting drive limit angle at some angle. But drive angle is not adjusting itself with respect to the rotation which added while creating joint ?
`
	PxRigidActor* actor0 = PxCreateStatic(*gPhysics, PxTransform(pos0), PxBoxGeometry(0.02, 0.02, 0.02), *gMaterial);
	PxRigidActor* actor1 = PxCreateDynamic(*gPhysics, PxTransform(pos1), PxBoxGeometry(0.5, 0.5, 0.5), *gMaterial, 1.0f);
	PxVec3 offset0 = PxVec3(0, 3, 0); // offset of connected actor0 from joint
	PxVec3 offset1 = PxVec3(0, -3, 0); // offset of connected actor1 from joint
	PxVec3 col1(1, 0, 0);								  
	PxVec3 col2(0, 1, 0);								
	PxVec3 col3 = col1.cross(col2).getNormalized();								 
	PxQuat checkAngle(PxPi/2 , PxVec3(1, 0, 0));
	PxMat33 Mat33(col1, col2, col3);
	PxQuat offsetRotation(Mat33);

// With Rotation 
	PxTransform offset0Trans(offset0,  offsetRotation * checkAngle);
	PxTransform offset1Trans(offset1, offsetRotation * checkAngle);

// Without Rotaion 
       // PxTransform offset0Trans(offset0);  
	//PxTransform offset1Trans(offset1);

	gd6joint = PxD6JointCreate(*gPhysics, actor0, offset0Trans, actor1, offset1Trans);
	gd6joint->setMotion(PxD6Axis::eTWIST, PxD6Motion::eLIMITED);
	gd6joint->setTwistLimit(PxJointAngularLimitPair(-PxPi/4, PxPi/2, PxSpring(0.0, 0.0)));
	gd6joint->setConstraintFlag(PxConstraintFlag::eDRIVE_LIMITS_ARE_FORCES, true);
	gd6joint->setConstraintFlag(PxConstraintFlag::eVISUALIZATION, true);
	gD6Drivelimit = PxD6JointDrive(10.0, 10.0, PX_MAX_F32, false);
	gd6joint->setDrive(PxD6Drive::eTWIST, gD6Drivelimit);
	gScene->addActor(*actor0);
	gScene->addActor(*actor1);
`

kstorey-nvidia, 2021-08-16:
Can you explain a little bit more exactly what you think is not working correctly? From the sample code, I can see that you've rotated the local joint frame by 90 degrees around the twist axis in both joint frames. What behavior did you expect this to change that didn't change?

I can  see that you've set some limits, but the spring parameters for the limits of 0 - this shouldn't mean the limits don't do anything so we can probably ignore them.

You are also setting a drive. The drive is trying to drive towards a relative transform between the joint frames which the application defines. As a the drive target position has not been set in this code, this target defaults to identity. This would mean that the joint will be trying to drive the parent and child frames to line up. If you rotate both frames by the same angle around the x-axis, the joint frames still line up, so the behavior of the drive shouldn't change. If you rotate only one of the frames, this would create an initial angular error that the drive would correct. This would be equivalent to not rotating the frames and instead setting a drive target, which will drive towards this offset between the joint frames.

AdilKhurshid, 2021-08-18:
As you noticed I'm rotating both the local frames by 90 degree around x axis so it should move the frames line up also 90 degree.
If I want to set the drive angle within the angular limit how can I set that ?
I can not use drive target at this place. 
[image]