NVIDIAGameWorks/PhysX #467: keeping vertex order in convex mesh

amir90, 2021-09-03:
I have a convex mesh with 8 vertices (which I know are on the convex hull) and I also need to access the vertex positions while simulating. 
It seems that the vertex order is not maintained with `PxConvexFlag::eCOMPUTE_CONVEX;`, 
I have tried to insert the hull polygons manually but I get the error:  Cooking::cookConvexMesh: user-provided convex mesh descriptor is invalid!

My code is:
```

	std::vector<PxHullPolygon> hullPolygons(6);
	hullPolygons[0].mNbVerts = 4;
        calculate_plane_eq(sheet_vertices[0], sheet_vertices[1], sheet_vertices[2], hullPolygons[0].mPlane);
	hullPolygons[1].mNbVerts = 4;
	calculate_plane_eq(sheet_vertices[1], sheet_vertices[5], sheet_vertices[6], hullPolygons[1].mPlane);
	hullPolygons[2].mNbVerts = 4;
	calculate_plane_eq(sheet_vertices[2], sheet_vertices[6], sheet_vertices[7], hullPolygons[2].mPlane);
	hullPolygons[3].mNbVerts = 4;
	calculate_plane_eq(sheet_vertices[0], sheet_vertices[3], sheet_vertices[7], hullPolygons[3].mPlane);
	hullPolygons[4].mNbVerts = 4;
	calculate_plane_eq(sheet_vertices[1], sheet_vertices[0], sheet_vertices[4], hullPolygons[4].mPlane);
	hullPolygons[5].mNbVerts = 4;
	calculate_plane_eq(sheet_vertices[4], sheet_vertices[7], sheet_vertices[6], hullPolygons[5].mPlane);

	PxConvexMeshDesc convexDesc;
	convexDesc.points.count = 8;
	convexDesc.points.stride = sizeof(PxVec3);
	convexDesc.points.data = &sheet_vertices;
	convexDesc.polygons.count = 6;
	convexDesc.polygons.stride = sizeof(PxHullPolygon);
	convexDesc.polygons.data = &hullPolygons;

```

I have two questions:

1. How do you insert a convex mesh using the hull polygons? And does this keep the vertex order?

2. What I really need is a dynamic deformable triangular mesh with `PxMeshPreprocessingFlag::eDISABLE_CLEAN_MESH`, but Physx allows it to be kinematic only. Given that this is a very small mesh, and performance is not an issue, is it possible to tell Physx to make it dynamic anyway?

thank you.

AlesBorovicka, 2021-09-10:
Hi,
 if you provide the PxConvexFlag::eCOMPUTE_CONVEX flag, then only the input vertices are used and a convex hull approximation is computed with these input vertices. The output is a subset of the input vertices.
If you dont provide this flag, then you need to pass the full geometry - vertices, indices and polygons. Then the vertex order is preserved as the full input data are used. From the snippet above it looks like you are missing the indices array?

As for the second question, yes PhysX only supports triangle meshes as kinematic bodies. 
Regards,
Ales

amir90, 2021-09-13:
Thanks, the indices array was indeed what I was missing.

closed 2021-09-13 by amir90