NVIDIAGameWorks/PhysX #471: Trouble with overlap

SevenGameMaker, 2021-09-12:
I cannot seem to get overlap working and any help would be greatly appreciated.

inside the object creation method i use this 

			if (actor->getType() == PxActorType::eRIGID_DYNAMIC)
			{
				//!convert the actor into a valid dynamic actor (we care about velocity and mass now :) )
				PxRigidDynamic* a2 = actor->is<PxRigidDynamic>();
				if (a2)
				{
#ifdef IGE_PHYSX_DEBUG_RENDER
					a2->setActorFlag(PxActorFlag::eVISUALIZATION, true);	//!if debug then enable visualizations
#endif

					a2->setAngularDamping(0.1f);									//!slow the object rotation down each frame by this much
					a2->setLinearDamping(0.1f);										//!slow the object velocity down each frame by this much
					a2->setLinearVelocity(PxVec3(0, 0, 0));							//!set the starting velocity to none
					a2->setMass(data.mass);
					a2->userData = (void*)data.userdata;							//!set the actor userdata (i.e. the container object ID number) 

					//!since we set the mass we need to update this
					PxRigidBodyExt::updateMassAndInertia(*a2, PxReal(data.mass));

					PxShape* shapes[10];
					u32 numShapes = a2->getNbShapes();
					PxU32 nShapes = a2->getShapes(shapes, numShapes > 9 ? 9 : numShapes);
					while (nShapes--)
					{
						shapes[nShapes]->userData = (void*)data.userdata;
						shapes[nShapes]->setFlag(PxShapeFlag::eSCENE_QUERY_SHAPE, true);

						if (data.istrigger)
						{
							shapes[nShapes]->setFlag(PxShapeFlag::eSIMULATION_SHAPE, false);
							shapes[nShapes]->setFlag(PxShapeFlag::eTRIGGER_SHAPE, true);
						}
						else
						{
							shapes[nShapes]->setFlag(PxShapeFlag::eTRIGGER_SHAPE, false);
							shapes[nShapes]->setFlag(PxShapeFlag::eSIMULATION_SHAPE, true);
						}
					}
				}
			}

now trying to perform an explosion i call

	void IGE_PhysX_World::doExplosion(vector3df pos, float distance, float force)
	{
		PxGeometry overlapShape  = PxSphereGeometry(distance);  // [in] shape to test for overlaps
		PxTransform shapePose = PxTransform(pos.X, pos.Y, pos.Z, PxQuat(PxIdentity));    // [in] initial shape pose (at distance=0)

		const PxU32		bufferSize = 256;        // [in] size of 'hitBuffer'
		PxOverlapHit	hitBuffer[bufferSize];  // [out] User provided buffer for results
		PxOverlapBuffer	buf(hitBuffer,256);

		int numHits = m_Scene->overlap(overlapShape, shapePose, buf);
		printf("numHits = %d\n", numHits);
	}

my results are that I receive either no hits or a single hit.
i only receive the one hit if my 'pos' variable is inside one of the dynamic objects.

Pierre-Terdiman, 2021-09-13:
This line is highly suspicious:

`PxGeometry overlapShape  = PxSphereGeometry(distance);  // [in] shape to test for overlaps
`

I would have to try & trace the code but it looks like you're creating a temporary PxSphereGeometry object and then only copy its base PxGeometry part to another temporary object, which is then used for the overlap test. I assume this is the source of your problems.

Generally speaking you don't need to go through the = operator or copy constructor for these things. A much simpler, safer and overall better way to write these lines would just be:

```
	PxSphereGeometry overlapShape(distance);  // [in] shape to test for overlaps
	PxTransform shapePose(pos.X, pos.Y, pos.Z, PxQuat(PxIdentity));    // [in] initial shape pose (at distance=0)

```

Please try this and see if it fixes the issue. If not, we can investigate further.

SevenGameMaker, 2021-09-13:
Worked like a charm. thank you for the help. also, the numHits was a bool value which I should have seen. using your suggestion and numTouches now performs as expected. thanks again for the help!

closed 2021-09-13 by PierreTerdiman

PierreTerdiman, 2021-09-20:
I updated the code so that the faulty line that created this problem won't compile now.