NVIDIAGameWorks/PhysX #472: Compile PhysX natively for Apple Silicon (M1)

niklas-ourmachinery, 2021-09-14:
Are there plans for supporting native compiles of PhysX on Apple Silicon (M1)?

Seems like in theory it shouldn't be too cumbersome, since there is already a `linux-aarch64` configuration as well as an `ios` one, but it might involve some fiddling with the settings.

niklas-ourmachinery, 2021-09-14:
I made some a quick hack to get it to compile (if anybody else is looking for this):

https://github.com/NVIDIAGameWorks/PhysX/commit/9fb98ab05294521cbeebab9ec7d18234e94eb219

(Note: This doesn't do things properly, it doesn't preserve the "x86_64" OS X build for example, does not create a new binary platform tag, adds PX_ARM to the list of 64-bit platforms, etc, etc. It's just a super quick hack to get things to compile.)

colincornaby, 2021-12-04:
Has this version been working well for people? Definitely not trying to put @niklas-ourmachinery on the spot, he just advertised it as a hack to get it compiling! But I'm having a few weird issues with collision detection that I'm not seeing on Intel. I'm trying to narrow it down specifically to PhysX and not some other weird side effect in the application. But so far everything points to PhysX just not handling some collisions on Apple Silicon.

colincornaby, 2021-12-04:
I went through the Physx SDK changes for Apple Silicon and found that some NEON optimizations were being built only for iOS. I carefully went through and added OSX defines alongside those iOS defines and so far the collision issues were fixed. I'll see if I can post something later.

Would certainly be easier if there was an official update for Apple Silicon though.

niklas-ourmachinery, 2021-12-04:
Oh great! Yes, I never tested this much so not surprising there are issues.

dpogue, 2022-06-04:
I think I've sorted out the M1 issues (and also the CMake build stuff), and put up a PR here: https://github.com/NVIDIAGameWorks/PhysX/pull/569