NVIDIAGameWorks/PhysX #473: Static scene guidelines?

Telliax, 2021-09-15:
Hi. I'm currently working on exporting a physics scene from external editor to physx. When I reconstruct the physx scene from the exported file, I generally create a static actor for each exported static collider, then I add corresponding physx shape to this actor and add it to the scene. This seems to be the most straight forward way of doing things, but I wonder if it is the recommended way? I struggle to find any tips or guidelines on this topic in documentation.
In particular, I am interested in:
1) Should I try to reduce the number of static actors? Basically is there a difference between 1 actor with 100 attached shapes (compound) and 100 actors with one shape each? In terms of how physx optimizes things internally.
2) Should I try to use aggregates, for logically connected static objects? Say, for tabletop and table legs.  
3) Any other general tips are welcome. 

P.S. I mainly interested in optimizing for scene query performance. I'm do not care much about the solver.

PierreTerdiman, 2021-09-22:
The scene-query (SQ) structures operate on shapes, so there is no difference between 1 actor with 100 shapes and 100 actors with 1 shape: for SQ this is the same.

Aggregates (PxAggregate) are not used for scene queries. So these won't help. But there is PxPruningStructure which does replicate the concept for SQ structures. However these are mainly to speed up insertion of new objects in the scene. It should have little effect on performance (basically you would end up with a tree of trees instead of a tree of shapes, but in theory there is no big difference in performance).

If you are dealing with triangle meshes, it can help to merge all of them together, then bin the triangles to grid cells, and then create a single triangle for each grid cell. This gives an homogeneous initial distribution that could give birth to better trees and better performance. Otherwise if the triangles are all over the place you can have N meshes overlapping each other - and in the worst case if you split meshes by material, they can all overlap the whole world and basically have the same initial bounding box.

Telliax, 2021-09-24:
Thanks for the explanation, it was very helpful!

closed 2021-09-24 by Telliax

PierreTerdiman, 2021-09-24:
Edited: "and then create a single triangle for each grid cell" => I meant a single mesh