NVIDIAGameWorks/PhysX #474: PhysX rotates dynamic objects by 180 degrees.

IceLuna, 2021-09-16:
Hi. 
The simulation runs fine if I run it on my PC (i5-9400f, GTX 1660 Ti) but when I run the same simulation on my laptop (i5-8250u, MX150) on the very first frame dynamic objects get rotated by 180 degrees around Z axis (points into the screen). It happens even if I lock rotation.

Before running the simulation
[image]
After
[image]

As you can see, the box and the barrel on the right are flipped.
Any ideas why is that happening?

kstorey-nvidia, 2021-09-17:
Can you please provide a PVD capture so we can confirm that the transforms are definitely different? Are you using GPU or CPU simulation?

IceLuna, 2021-09-17:
I'm using CPU simulation. PVD shows that rotation is 0...
[image]

Apparently, it wasn't PhysX problem. For some reason, glm::eulerAngles returns 180 sometimes... 
[image]

Thanks for the help!

closed 2021-09-17 by IceLuna