NVIDIAGameWorks/PhysX #475: Direction dependent forces

asdasdasdasdasdf, 2021-09-17:
This is a question about how to use PhysX. Not a bug report.

I have several circumstances where I want to prevent an object moving by supplying up to a maximum force. I typically achieve this by applying a damping joint with very very high damping coefficient and the appropriate force limit. I think you could equivalently apply a  spring  joint with very very high spring coefficient or possibly there's a way to do it with locked joints. The key is that
A) there's a maximum force
B) the maximum force is dependent on direction

Sometimes my case is as simple as saying the object cannot go backwards. In physx 3 I could achieve this with a customer joint and in physx4 I believe you can set the min and max force independently so a D6 joint has everything I need.

However sometimes I want a different force in orthogonal directions. e.g. it resists movement along X with up to 10N of force and movement along Y with up to 20N. The two ways I've attempted this in the past are:
A) try to anticipate the movement ahead of time, calculate resistance to that movement and then apply a joint in the anticipated axis
B) apply orthogonal joints with the respective limits

option A is limited by how good my estimated movement axis is. If it's a very good guess the approach is fine. If it's a bad guess it creates weird behavior as you'd imagine. It's particularly complex to anticipate motion since the motion is itself dependent on the joints in question.

option B is bad in a different way. It works fine if the force happens to be precisely along one of the primary axes but it makes movement difficult along any other axis. If I apply a force along (1, 10) then ideally I'd want a resistance force in the direction (-1,-10) but instead I get two resistance forces in the directions (-1, 0) and (0, -1). In this case if my two limits were reasonably similar then that would probably mean the X component joint would  easily oppose movement in the X direction. I think this problem is somewhat similar to the simulation of friction where I'm lead to believe that a naive solution will be correct in the unit axes and incorrect in other axes (something about a friction box vs a friction circle? Not an area in which I'm well read)

Ideally, although I've never tried this, I'd like to complicate the problem even further by allowing for a complex relationship between the direction and the force (for the sake of argument let's say a linearly interpolated lookup into a table). Option B wouldn't work at all in this case I think and option A doesn't get any easier.

The above is the problem I'd like a solution to. Below is my ideas on what might be viable.

It occurs to me that this problem is not that dissimilar to a distance joint (the functionality of which I believe is now built into D6 joints as well as of physx 4). The way that it solves the problem, as far as I can tell, is that it's effectively using option A except that the calculation of the direction of motion to oppose is done in the constraint shader rather than option A which is outside the simulation tick.

How frequently is this called? Is it once per position iteration or once per simulation tick? Would it be viable to set up a distance joint with very very high damping and configure its max force based on the calculated position vector in the constraint shader? Are there good theoretical reasons why I shouldn't do this? In the past I've implemented ideas that seem to work but violate unwritten rules of physics simulation (e.g. redundant joints connecting the same bodies) so I'd like to know if this idea is theoretically sound before attempting it.

Thanks

asdasdasdasdasdf, 2021-09-20:
I investigated a little and it seems like the constraint shaders are only executed once per physx tick irrespective of position iterations. That makes me think my idea isn't going to work. Possibly it's not a problem for distance joints because the direction can be approximated to be constant over a single tick?

Any suggestions as to other solutions to my problem would be welcome.

kstorey-nvidia, 2021-09-20:
Agree that this is a problem. It can be solved, but this is not something that can be easily solved with the solver currently in PhysX 4. 

As you identified, we have two options currently available: multiple independent constraints, or trying to approximate the asymmetric cone using a single constraint that samples on the closest point on the surface of the cone to the current state. The first option produces a robust limit solution, although this provides you with a force limit pyramid rather than a smooth asymmetric cone. The cone would be fully contained within the pyramid and would coincide along the principal axes. The second option gives you a closer match to the asymmetric force limit cone you want, but it can fail if the motion changes sufficiently that the chosen axis is no longer a good approximation. This approach works OK - this is how the swing limits are implemented in spherical/D6 joints - but it is not perfect.

One option that might be available is to use a combination of both the pyramid and cone limits. There is nothing stopping you from having both and this means that the pyramid can act as a back-stop to catch the cases where the cone approximation failed. It won't be perfect, but it can work well.

In PhysX 5, we have a new approach for simulating friction cones that converges to the exact analytical cone for our friction constraints. We haven't extended the joints to support this approach, but it is something we hope to do in the future.

asdasdasdasdasdf, 2021-09-20:
Combining the two approaches is an interesting idea. Wouldn't that be quite likely to create redundant constraints though? i.e. if I determine that my primary axis for the cone is the x axis and I already have a joint acting in the x axis won't that cause problems?

The solutions you're talking about in PhysX 5 sound exciting. I hope I'll be able to utilize them at some point.

asdasdasdasdasdf, 2021-09-20:
oh also wouldn't I effectively get up to twice the force from the cone+pyramid?

kstorey-nvidia, 2021-09-20:
Yes, you're right. As this relates to a force limit and not just a hard constraint, it would end up yielding twice the force in some cases so not a good solution for you :(. We do have cases where developers are using the pyramid limits on D6 joints as a back-stop for the cone limits. Unfortunately, this is only going to yield acceptable results if they are hard limits.

asdasdasdasdasdf, 2021-09-20:
ok. thanks for your advice nonetheless.

closed 2021-09-20 by asdasdasdasdasdf