NVIDIAGameWorks/PhysX #476: Kinematic vs Dynamic collision problem

marcell-baranyai, 2021-09-21:
Hello,
I've been playing around for a while and can't get my head around how to set up a kinematic body "properly".
- The dynamic bodies on moving kinematic tend to be jumpy, even if it's moving slow.
- Also when it rotates the dynamic bodies are drifting.
- Occasionally when there are many dynamic boxes stacking on top of a kinematic the whole island sleep in mid-air.

I use PxRigidDynamic::setKinematicTarget()

Reference videos:
https://youtu.be/OvhArULJLLU
https://youtu.be/jPxyoAZEbS4

Here are some captures, just in case.
[KinematicCapture.zip](https://github.com/NVIDIAGameWorks/PhysX/files/7206153/KinematicCapture.zip)

thanks for any help and hints.

PierreTerdiman, 2021-09-22:
I didn't look at the capture yet but try increasing the contact offset first.

marcell-baranyai, 2021-09-22:
Unfortunately modifying the contact offset doesn't help. Increasing the value even escalates the effect.
I tried various settings between 0.001 and 1, and with higher values the body was bouncing off the platform. :(

kstorey-nvidia, 2021-09-22:
I took a look at the PVD capture and it looks like the issue is that the kinematic targets are not being set consistently.

If you look at the kinematic_move.pvd capture from an arbitrary frame (I'm looking at frame 185), we get the following pattern:

185 - velocity of kinematic: 0
186 - velocity of kinematic: 0, -0.9138, 0
187 - velocity of kinematic: 0, -1.593, 0
188 - velocity of kinematic: 0
189 - velocity of kinematic: 0, -1.479, 0
190 - velocity of kinematic: 0
191 - velocity of kinematic: 0, -1.347, 0
etc.

This is creating a vibrating effect in the simulation because the platform is not moving smoothly but instead moving fast, stop, fast, stop etc.

I wonder if you have a mismatch between the simulation rate and the rate at which you update your game engine/render? You need to set kinematic targets each simulation step and you would probably need to be interpolating the targets you set to PhysX if you are running multiple steps per rendered frame if you want to achieve smooth motion.

marcell-baranyai, 2021-09-22:
Thank you @kstorey-nvidia for pointing that out.
The situation was exactly what you suggested. Interpolating the target between physics sub-steps was the way to go. :)

closed 2021-09-22 by marcell-baranyai