NVIDIAGameWorks/PhysX #477: Contact target velocity affected by small position differences

kermado, 2021-09-28:
My scene is composed of two straight conveyors connected by a curved conveyor. I'm setting the contact target velocity to move an object along the conveyors. The object appears to be getting skewed after passing over the curved conveyor. Looking at the contact impulses, I can see that as the COM of the object has passed from over the first straight conveyor to over the curved conveyor, the impulses for the contact points furthest over the curved conveyor are zero.

[image]

The positions of the conveyors are determined by a bunch of single-precision matrix multiplications and so there may be tiny differences in the height of the conveyor geometry. I have been able to create a short sample that demonstrates this. Increasing or decreasing the height of the curved conveyor by 0.1mm above the other conveyors causes the object to become skewed.

[image]

It looks like the small height differences are causing the normal force (and hence the friction force) to be zero. Is there any way round this? Increasing the `frictionOffsetThreshold` seemed to have no effect.

The source code for the sample is here: https://gist.github.com/kermado/5abfbcf9f97c788fb56a35c21e7c9b32

Thanks

kstorey-nvidia, 2021-10-04:
Similar issues have been discussed in the past and they do not appear to be a fault in the dynamics. When we have a set of contacts supporting a rigid body, some of those contacts can apply zero impulse if the other contacts are already fully supporting the body. Increasing friction offset won't help in this case. Friction offset increases the separation distance at which we begin to factor in frictional forces, but friction is still always proportional to normal force so, if normal force is zero, so will friction force.

One thing that might work is if you reject the contacts at the seam between the two shapes in your contact modification. this would avoid the case where the box is fully supported by one of the conveyors, causing the other to have insufficient frictional effect. This might resolve this skewing effect.

There are other options, which are more invasive changes to the solver, that you could perhaps add but they will be physically incorrect (e.g. forcing a minimum amount of normal force for friction for your contacts). I'd recommend against trying things like this unless you have a pretty good handle on how the internals of the solver works, though.

kermado, 2021-10-04:
Thanks for the explanation Kier. This all makes sense and is pretty much what I expected. I noticed though that ODE seems to be more forgiving in this case. Could this be due to the CFM parameter? If so, is there an equivalent parameter in PhysX?

kstorey-nvidia, 2021-10-05:
We don't have an equivalent of CFM in PhysX outside of it being used in the articulation solver to soften some of the constraints to avoid degeneracies.

Adding a relaxation coefficient or a CFM term would soften the response from the solver, making it take more iterations/more contacts to converge on an answer. This would most likely result in all contacts applying having some normal force in this case, which would get you the frictional behavior you are looking for. It would make stacks/piles require more iterations though so there are trade-offs.

As mentioned, there isn't an equivalent variable for you to tweak so this would be something you would need to add. Alternatively, you can reject/limit forces on the contacts in contact modification to get a similar effect. You can definitely use contact modification to implement softer collisions.

closed 2021-10-05 by kermado