NVIDIAGameWorks/PhysX #479: Porting PhysX to FreeBSD

Esentiel, 2021-10-01:
Hello!

I recently moved to FreeBSD on my home desktop and to be able to work on my pet project(that depends on PhysX) I ported it to FreeBSD. I made patch that contains necessary changes to compile PhysX library on FreeBSD. Still I didn't do anything about Samples and Snippets, simply because I didn't use it.
Now I think if you guys can/want extend portability for PhysX 4 and maintain FreeBSD support, I can port samples, snippets as well and submit a proper pull request with changes.

Cheers

Martinfx, 2024-07-05:
some progress  ? is year 2024

preist-nvidia, 2024-07-05:
Hi both - we will not update supported platforms for PhysX 4 anymore, and for PhysX 5 we are currently not planning to expand the supported platforms.

closed 2024-07-05 by preist-nvidia