NVIDIAGameWorks/PhysX #480: Crash in 'Sc::Scene::unregisterInteractions'

GregSlazinski, 2021-10-03:
Hello,
One of users in my engine is getting crashes happening in PhysX version 4.1.1.27006925
Sc::Scene::unregisterInteractions
[image]
I've seen some other programmer had similar crashes related to 'unregisterInteractions'
https://forums.developer.nvidia.com/t/gym-simulate-hanging-on-pthread-cond-wait-in-physx-wait-unsigned-int/169149/4
Do you have any idea what's the problem and how to fix this?
Or is this fixed already in 4.1.2? We're on 4.1.1
The user said crashes happen in game during fights with monsters - at that time ragdolls and aggregates are turned on.
Please help, thanks

GregSlazinski, 2021-10-03:
I was able to reproduce the problem, and attaching callstack from debug:

[image]

[image]

[image]

PierreTerdiman, 2021-10-03:
The scene array index is undefined. This should not happen. I suspect it comes from a duplicate pair, that could probably lead to this kind of issues. So that would be a bug in the AABB manager higher up the pipeline. I don't immediately see a fix for something like this in the file's history, but that history is quite large so maybe I missed it. Will ask a few other people.

Meanwhile some defensive code should avoid the crash:
```

void Sc::Scene::unregisterInteraction(Interaction* interaction)
{
	const InteractionType::Enum type = interaction->getType();
	const PxU32 sceneArrayIndex = interaction->getInteractionId();
	**if(sceneArrayIndex==PX_INVALID_INTERACTION_SCENE_ID)
		return;**
...

```

Add the last 2 lines.

GregSlazinski, 2021-10-06:
Thanks, if I do that, I'm getting crashes here:
[image]
[image]
[image]
[image]

mastercoms, 2022-10-11:
I'm getting this rarely but often enough that it's a disruptive problem. Using [PhysX_4.1.2.29882248](https://github.com/NVIDIAGameWorks/PhysX/commit/c3d5537bdebd6f5cd82fcaf87474b838fe6fd5fa). 

This is the task queue for the crash:

[image]

Here's the AABB Manager:

[image]

ABP:

[image]

I have the full dump file, let me know where I can send it if needed.