NVIDIAGameWorks/PhysX #491: Kinematic to dynamic actor - Uncertain about how PhysX resolves collisions

amir90, 2021-10-31:
Hello, 

I have a scenario where I need a simple deformable convex mesh for simulation. As Physx does not allow changing the vertices of of a convex mesh, I recreate it each frame with the required geometry. To handle collisions with the environment, my attempted solution is to use the `PxPairFlag::eNOTIFY_TOUCH_FOUND`, and to turn on the shape's simulation flag and let PhysX resolve the collision. When the collision is resolved I would turn off the simulation flag and continue.

However, I am not sure what the "collision resolved" flag should be. Is there `eNOTIFY_PENETRATION_FOUND` flag?
Initially, I assumed PhysX resolves the collision within 1 frame, but that does not seem to always be true, even if I set `MaxPenetrationVelocity `to be the maximum (see attached PVD capture, where the actor penetrates the environment after the update, and is turned to a simulation object, but the penetration is not resolved in the next frame). 

[image]

I would appreciate your help. Thank you!

kstorey-nvidia, 2021-11-02:
PhysX can't guarantee it will resolve all penetrations in a single frame, so you definitely can't guarantee that. What you could potentially do is use a force thresholds/examine the applied forces to determine when the solver stops applying forces. You could alternatively just examine the set of contacts to determine when separation becomes >= 0.

amir90, 2021-11-02:
Ok, Thank for the reply! examining the separation sounds promising, I will try.