NVIDIAGameWorks/PhysX #497: Sweep normals on edges

nikita-one, 2021-11-19:
Hi. I have noticed, that if geometry (in my case a capsule) hits an edge during scene sweep, the blocking hit normal seems to be a product of some sort of interpolation between the normals of faces that form given edge.
So my questions are: 
1) Is there a way to get actual normals of both surfaces without doing additional queries?
2) If not is there a reliable way to get those normal with additional queries? I tried doing raycasts with small offsets from original hit point but this approach feels... shaky.

Contex: I perform a downwards sweep in order to detect floor surface. So when hitting edge, I would like to check both surfaces to see, which one is closer to horizontal plane. Basically I am looking for a way to reliably get floor normal (blue) instead of edge "normal" (red):
[image]

PierreTerdiman, 2021-11-19:
The internal code is tweaked so that it returns the blue polygon as the closest triangle index. So you should get the normal you want by just computing the normal of returned triangle.

nikita-one, 2021-11-19:
Thanks, that's good to know. And if I'll ever want to get the another one? The wall normal, which points left on the image above?

nikita-one, 2021-11-19:
Also I'm not sure what do you mean by "closest", can you please elaborate on that? Closest relative to what?

PierreTerdiman, 2021-11-19:
I just mean that when you do a sweep we return the "closest hit", with the various hit parameters like position / normal / etc in the sweep hit structure. In the case you describe both the horizontal and vertical polygons have the same impact distance but the code is tweaked so that the horizontal one is seen as "closest".

PierreTerdiman, 2021-11-19:
If you need the other one, there is no easy way to do that right now. I think the only option would be to move the swept shape to the point of impact and then perform an extra overlap test there with a slightly increased shape volume to retrieve touching triangles. Then you can analyze the normals of returned triangles in your own code.

PierreTerdiman, 2021-11-19:
Another option would be a "sweep multiple" query that would return multiple hits for a single triangle mesh but I think we don't support this (only for raycasts).

nikita-one, 2021-11-19:
> tweaked so that the horizontal one is seen as "closest"

Ah, I see. So the direction of the sweep plays no role in this. A more "horizontal" triangle is always prioritized, even if I were to sweep from left to right, correct?

PierreTerdiman, 2021-11-19:
No the direction plays a role. If you would sweep from left to right we would return the wall triangle as "closest".

PierreTerdiman, 2021-11-19:
Yeah sorry I wasn't clear: we internally apply a "tweak" to computed distances based on the dot product between the touched triangle's normal and the sweep direction.

nikita-one, 2021-11-19:
Got it, thanks for the help!

closed 2021-11-19 by nikita-one

PierreTerdiman, 2021-11-19:
Actually I forgot something; this internal tweak seems to be only implemented for triangle meshes. So if you hit a box or a convex primitive this won't be applied :(

nikita-one, 2021-11-19:
I see, thanks, I'll keep that in mind. We might be fine with using triangle meshes instead of boxes, if it comes to it