NVIDIAGameWorks/PhysX #499: Spring Contacts

alanjfs, 2021-11-21:
Hi all,

I'd like to implement spring contacts with PhysX, but am struggling.

I found PxContactModifyCallback, whose documentation [refers to "soft" contacts](https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/Manual/AdvancedCollisionDetection.html) which I suspect refers to this.

https://user-images.githubusercontent.com/47274066/142777369-5e97641e-89a1-46e2-8b50-2d12b0805980.mp4

Which... sure, one could argue is "soft" but not exactly "springy". How would I go about making these springy?

I'm using PhysX 4.1.

alanjfs, 2021-12-06:
Still banging my head against this, it doesn't seem possible with PhysX? :( I've been spoiled working with ODE, where soft contacts can easily be achieved via CFM and ERP.

My understanding of the D6 joint limits is that they are effectively solved as collisions against the limit, but with a `stiffness` and `damping`  value. So part of me would still like to believe it *is* somehow possible.

Any pointers on this? Is it a fruitless endeavour?

kstorey-nvidia, 2021-12-06:
This is not something that we have support for in PhysX 4. It is, however, likely to be something that we are interested in adding official support for in the future.

The D6 soft joint limits are not too dissimilar from how we would implement soft contacts.

alanjfs, 2021-12-06:
Thanks for confirming @kstorey-nvidia. Any pointers as to where one would start poking around if one wanted to try and apply the soft joint limits to contacts?

kstorey-nvidia, 2021-12-06:
There are quite a few places, so best suggestion would be to constrain the set of features you want to try and use.

Assuming you only want to work with rigid bodies and not articulations, and you want to use the PGS and not TGS solver, the first suggestion would be to temporarily disable the batched solver controlled with this define:

#define DY_BATCH_CONSTRAINTS 1

Then you can take a look at

alanjfs, 2021-12-06:
Thank you! Diving into this.