NVIDIAGameWorks/PhysX #503: CCD contact normals is reversed and faceIndex is invalid

pkusilence, 2021-11-24:
I am testing CCD of PhysX 3.4 with Unreal Engine 4.26. I found that the normals generated by CCD is not the same as the discrete version (DCD) in contact modification. As claimed in https://gameworksdocs.nvidia.com/PhysX/3.4/PhysXAPI/files/structPxContactPairPoint.html : "The normal direction points FROM the second shape TO the first shape." It is true for DCD but not keeped in CCD. Besides, faceIndex in CCD is not valid but DCD works well too.

How could I fix these problems?

kstorey-nvidia, 2021-11-24:
The issue with face indices was recently reported by Unity. CCD contact reporting missed reporting face indices, and this is slated to be fixed soon and then included in the next PhysX release. We will also verify the direction of reported normals is also consistent and fix if required.

If you would like to fix these issues yourself and submit a pull request, feel free to do so. Most of the CCD code is in PxsCCD.cpp. Contact reporting is processed in postCCDAdvance method.

pkusilence, 2021-11-26:
A fix is summitted, please take a look

pkusilence, 2021-11-29:
An other problem: The inv Mass/Inertia scale in CCD modification is not working.

kstorey-nvidia, 2021-11-29:
Thanks. Yes, I see the issue. InvInertiaScale wouldn't work because the response is entirely linear, unless you enable the CCD_ANGULAR_IMPULSE define. This was known to cause some behavioral issues so it was disabled.

The missing use of invMassScale is an oversight. The CCD code does support dominance, which offers the ability to make one body appear infinite mass for the duration of the collision, but it wasn't adapted to support invMassScale, which was a later addition to the contact callbacks.

In bool PxsCCDPair::sweepAdvanceToToi(PxReal dt, bool clipTrajectoryToToi)

you should be easily able to add in support for invMassScale by modifying these lines:

PxReal dom0 = mCm->getDominance0();
PxReal dom1 = mCm->getDominance1();

and adding in the invMassScale terms to the respective dom0 and dom1 values. This will involve adding the respective invMassScale values to PxsCCDPair and copying them over from patch->mMassModification.mInvMassScale0 and patch->mMassModification.mInvMassScale1 after the call to runCCDModifiableContact.

We will add this fix into the latest PhysX development trunk so the fix should become available with the next release.

pkusilence, 2021-11-29:
Thanks for reply. I will take a try. 

I am still wondering wheather PhysX works like that CCD pipeline go through broadphase & sweep & collision solver, and then go into DCD pipeline? Since it is observed that after all the CCD collision stuff, a discrete collision is still found and the DCD contact modification is excuted.

closed 2021-12-02 by pkusilence