NVIDIAGameWorks/PhysX #508: Do you know how to use PxSimulationEventCallback?

MoonHJ98, 2021-12-10:
I want to receive events on crash, but I'm not sure how to use PxSimulatoinEventCallback.

     class ContactCallBack : public PxSimulationEventCallback
     {
	     // PxSimulationEventCallback을(를) 통해 상속됨
	     virtual void onConstraintBreak(PxConstraintInfo * constraints, PxU32 count) override;
	     virtual void onWake(PxActor ** actors, PxU32 count) override;
	     virtual void onSleep(PxActor ** actors, PxU32 count) override;
	     virtual void onContact(const PxContactPairHeader & pairHeader, const PxContactPair * pairs, PxU32 nbPairs) override;
	     virtual void onTrigger(PxTriggerPair * pairs, PxU32 count) override;
	     virtual void onAdvance(const PxRigidBody * const * bodyBuffer, const PxTransform * poseBuffer, const PxU32 count) 
             override;
     };

I created a callback class and put it in the scene

     HRESULT PhysXManager::Initialize()
     {
          '''''

          PxSceneDesc sceneDesc(physics->getTolerancesScale());
          sceneDesc.gravity = PxVec3(0.f, -9.81f, 0.f);
          dispatcher = PxDefaultCpuDispatcherCreate(2);
          sceneDesc.cpuDispatcher = dispatcher;
          sceneDesc.filterShader = PxDefaultSimulationFilterShader;
          sceneDesc.simulationEventCallback = new ContactCallBack();
          scene = physics->createScene(sceneDesc);

          '''''
     };

Then shouldn't onContact be called when shapes collide?

closed 2021-12-10 by MoonHJ98

fbz1233333, 2024-11-05:
so .. why?

vreutskyy, 2024-11-05:
You need to request contact information per pair in the filter shader.
https://nvidia-omniverse.github.io/PhysX/physx/5.4.2/docs/AdvancedCollisionDetection.html#contact-reporting