NVIDIAGameWorks/PhysX #514: No apparent way to make CCT collide with static rigid bodies

Moi78, 2021-12-19:
I'm surprised that this is not possible, well at least it seems impossible, and the docs aren't helpful at all and doesn't detail that. It seems to be pretty random. In some cases, there are collisions in others there are no collisions, the declaration of the rigid body, CCT etc... are the same in both cases, but depending on the code following it (which is btw not related to physics in any kind) theses collisions are enable or not. This is really weird. 

I think I don't need to explain how confusing it is....

PierreTerdiman, 2022-01-03:
I'm afraid I don't follow the question and I don't understand the problem. CCTs do collide with static rigid bodies, as shown in several samples (SampleNorthPole, SampleCustomGravity, SampleBridges, etc).

Moi78, 2022-01-03:
Well, it may be a linking problem nor a compiling problem caused by me. Note that it appears on both linux and windows and that I used the provided build system (underlying on cmake) provided in the repo. Interestingly, every samples works like a charm but when I link it and use it in my project, it doesn't work anymore. I mean, it effectively pushes other dynamic actors but it does not collide with static objects.

I don't know if I can do or send anything that can be helpful here.

PierreTerdiman, 2022-01-03:
Perhaps try to use the debug visualization to make sure that the static meshes are where you expect them to be, and maybe double check the triangle winding for these meshes. Not sure what I can do without a repro otherwise.

Moi78, 2022-01-03:
In the PVD, the CCT Capsule is where I placed it in my scene and as I move, it moves through static objects as if they were not there.