NVIDIAGameWorks/PhysX #515: Different Simulation Behaviour in Release and Debug build

matangpanchal, 2022-01-05:
Hi,
I have around 20 Rigid Dynamics Actors having a convex Mesh inside a triangle static mesh. according to my requirements I scale / Translate  shapes. Things are working as they should in debug build, but when i use release build I am getting performance boost but I think contacts and other physx related things are misbehaving or differently behaving so that I am not getting same result as debug build. 
So any chances are there that physx setup takes different options for solving and other internal parameters across different builds ?
Please help.
Regards,

PierreTerdiman, 2022-01-06:
If it's "misbehaving" then there might be an issue/bug to investigate.

But just "differently behaving" is normal between debug & release builds. The compiler reorganizes instructions etc as part of its optimization passes and the result is that there's no determinism between debug & release - things will behave slightly differently, that's perfectly normal.

jankrassnigg, 2022-01-06:
Do you use fixed time stepping? If not, and your timestep is drastically lower in Release than in Debug, that can also explain different behavior.

matangpanchal, 2022-01-07:
> Do you use fixed time stepping? If not, and your timestep is drastically lower in Release than in Debug, that can also explain different behavior.

I think my problem is something as you are describing... But I am sorry I am not able to figuring out am i using Fixed time stamping or variable.

Here is a brief what I am doing, please guide me.
My simulation is not needed to be visualize, so I am not rendering it any where. 
So after initializing scene & physics, I am looping step physics in a while loop
i.e 
while (true)
{
 stepPhysics(false);
} 

then in stepPhysics I am doing 
void stepPhysics(bool interactive)
{
 PxReal deltaTime = 1.0f/500.0f;
gScene->simulate(deltaTime);
gScene->fetchResults(true);

///Rest of all of my alogrithms run below this 
}

I think this a  fix time stepping, but not sure.. If this is not fix time stepping then what can be the fixed time stepping ?

matangpanchal, 2022-01-07:
> If it's "misbehaving" then there might be an issue/bug to investigate.
>
> But just "differently behaving" is normal between debug & release builds. The compiler reorganizes instructions etc as part of [...]

Actully my whole results gets changed in debug and release build, so I am also confused that is this a misbehaviour or different behaviour.

kstorey-nvidia, 2022-01-10:
Do you get similar results with checked PhysX builds compared to debug builds or release builds? If checked produces results that are more like the debug build, are there any warnings being emitted from PhysX? Debug and checked build sanitize API calls and reject invalid inputs with error messages, while release builds skip these validation checks for performance reasons.

If checked and debug differ, it would be helpful to get PVD captures of debug and checked build so that we can see the behavioral difference to identify if it looks like a bug or not.