NVIDIAGameWorks/PhysX #517: PhysX 5.0 - Available in 2022 ?

SparkyTD, 2022-01-10:
On the [PhysX website](https://developer.nvidia.com/physx-sdk) it is stated that the new SDK can be downloaded as part of an "Open Beta" in the NVIDIA Omniverse platform. I have downloaded the Omniverse app, but I haven't been able to find a download link to a C++ SDK. The way this is phrased on the website suggests that the SDK is downloadable in the open beta, which might be misleading. Downloading a pre-made sample app that *uses* PhysX 5 does not count as "downloading PhysX 5", and is completely irrelevant as far as developers are concerned.

Someone please tell me where I can sign up for this open beta, where I can download the C++ headers, static- and shared library files, or if that's not possible at this time, then please remove the misleading text from the website as soon as possible.

[image]

RobertMtx, 2022-04-14:
I agree 100%. Very misleading. Apparently it is a gimmick to get more devs to installs Omniverse? I fell for it. Where is our PhysX 5 SDK promised at the NVidia website?

vinjn, 2022-11-08:
https://github.com/NVIDIA-Omniverse/PhysX