NVIDIAGameWorks/PhysX #521: Crash when character controller moves while standing on actor that gets destroyed

jankrassnigg, 2022-01-18:
The title pretty much says it all. When a physics object gets deleted (in my tests it's a PxArticulation) on which a character controller is standing, and the CC also moves during that frame, then PhysX crashes.

I was wondering whether this is a timing issue, ie whether I have to update the CC at some other time, ie. before or after the scene simulation, to prevent this issue.

Anyone got any insights?

PierreTerdiman, 2022-01-19:
It's probably because the character controller code hasn't been updated when articulations got introduced. Gut feeling is that the code here should be updated to support articulations:

https://github.com/NVIDIAGameWorks/PhysX/blob/4.1/physx/source/physxcharacterkinematic/src/CctCharacterControllerManager.cpp#L215

jankrassnigg, 2022-01-19:
Good hint, thank you, so at least that means I could fix it on my end. And of course someone can fix it in the soon to be released PhysX 5 ;-)

PierreTerdiman, 2022-01-24:
FWIW I created a repro for this bug and indeed, the fix is simply to add support for articulation links in the file above. Something like this:

```
	const PxType type = observed->getConcreteType();

	if(type!=PxConcreteType:: eRIGID_DYNAMIC && type!=PxConcreteType:: eRIGID_STATIC && type!=PxConcreteType::eSHAPE && type!=PxConcreteType::eARTICULATION_LINK)
		return;

```

jankrassnigg, 2022-01-24:
Thanks for checking!

closed 2022-01-24 by jankrassnigg