NVIDIAGameWorks/PhysX #522: Building for Visual Studio 2022?

nothke, 2022-01-19:
Attempting to configure with 2019 option reports missing VS 2019. I'm looking at cmake files but it's utterly confusing. Anyone has any idea how to add a 2022 build?

I can't believe that I just updated to 2022 a couple of days ago to find out you can only build PhysX up to 2019 :D I don't want to reinstall 2019 just for this.

Zeblote, 2022-01-20:
Adding support for VS 2022 yourself is quite easy. I have done so here. https://github.com/Zeblote/PhysX/commit/00180480a25c8ac6172d3703a4939af8dc06812b

nothke, 2022-01-20:
Ooh, that's great! Thanks a lot!

youss2017, 2022-04-29:
use this branch for vs 2022 git clone -b brickadia-4.1.2 https://github.com/Zeblote/PhysX

Zeblote, 2022-04-30:
Worth noting that my branch is specialized for brickadia, it might not be ideal for all uses cases, for example shared shapes are broken.

lotusblue963, 2022-05-17:
I tried using it but it's still not building it

slackchen, 2022-06-29:
I've do it for myself too: https://github.com/NVIDIAGameWorks/PhysX/pull/577
lol~

Pacheco95, 2022-10-16:
@Zeblote thaks. It works perfectly for me.