NVIDIAGameWorks/PhysX #527: Including PhysX 4.1.2 in a CMake Project

bilek993, 2022-02-05:
Hello,
Based on issue #249, I'm trying to add PhysX into the new CMake Project. The project with PhysX subproject builds and produces `.dll` and `.lib` files, but in the final steps, the linker fails to link PhysX libraries into the main project. Here is the log output:

```
LINK Pass 1: command "C:\PROGRA~2\MICROS~2\2019\COMMUN~1\VC\Tools\MSVC\1429~1.301\bin\Hostx86\x64\link.exe /nologo @CMakeFiles\physics_testing.dir\objects1.rsp /out:physics_testing.exe /implib:physics_testing.lib /pdb:C:\Users\jbili\Documents\Cpp_Projects\physics_testing\cmake-build-debug\physics_testing.pdb /version:0.0 /machine:x64 /debug /INCREMENTAL /subsystem:console externals\physx\bin\win.x86_64.vc142.mt\debug\PhysXExtensions_static.lib externals\physx\bin\win.x86_64.vc142.mt\debug\PhysX_static.lib externals\physx\bin\win.x86_64.vc142.mt\debug\PhysXPvdSDK_static.lib externals\physx\bin\win.x86_64.vc142.mt\debug\PhysXVehicle_static.lib externals\physx\bin\win.x86_64.vc142.mt\debug\PhysXCharacterKinematic_static.lib externals\physx\bin\win.x86_64.vc142.mt\debug\PhysXCooking_static.lib externals\physx\bin\win.x86_64.vc142.mt\debug\PhysXCommon_static.lib externals\physx\bin\win.x86_64.vc142.mt\debug\PhysXFoundation_static.lib externals\physx\bin\win.x86_64.vc142.mt\debug\PhysXPvdSDK_static.lib externals\physx\bin\win.x86_64.vc142.mt\debug\PhysXFoundation_static.lib kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib /MANIFEST /MANIFESTFILE:CMakeFiles\physics_testing.dir/intermediate.manifest CMakeFiles\physics_testing.dir/manifest.res" failed (exit code 1120) with the following output:
LINK : warning LNK4217: symbol 'PxCreateFoundation' defined in 'PhysXFoundation_static.lib(PsFoundation.cpp.obj)' is imported by 'main.cpp.obj' in function 'main'
main.cpp.obj : error LNK2019: unresolved external symbol __imp_PxRegisterArticulations referenced in function "class physx::PxPhysics * __cdecl PxCreatePhysics(unsigned int,class physx::PxFoundation &,class physx::PxTolerancesScale const &,bool,class physx::PxPvd *)" (?PxCreatePhysics@@YAPEAVPxPhysics@physx@@IAEAVPxFoundation@2@AEBVPxTolerancesScale@2@_NPEAVPxPvd@2@@Z)
main.cpp.obj : error LNK2019: unresolved external symbol __imp_PxRegisterArticulationsReducedCoordinate referenced in function "class physx::PxPhysics * __cdecl PxCreatePhysics(unsigned int,class physx::PxFoundation &,class physx::PxTolerancesScale const &,bool,class physx::PxPvd *)" (?PxCreatePhysics@@YAPEAVPxPhysics@physx@@IAEAVPxFoundation@2@AEBVPxTolerancesScale@2@_NPEAVPxPvd@2@@Z)
main.cpp.obj : error LNK2019: unresolved external symbol __imp_PxRegisterHeightFields referenced in function "class physx::PxPhysics * __cdecl PxCreatePhysics(unsigned int,class physx::PxFoundation &,class physx::PxTolerancesScale const &,bool,class physx::PxPvd *)" (?PxCreatePhysics@@YAPEAVPxPhysics@physx@@IAEAVPxFoundation@2@AEBVPxTolerancesScale@2@_NPEAVPxPvd@2@@Z)
main.cpp.obj : error LNK2019: unresolved external symbol __imp_PxCreateBasePhysics referenced in function "class physx::PxPhysics * __cdecl PxCreatePhysics(unsigned int,class physx::PxFoundation &,class physx::PxTolerancesScale const &,bool,class physx::PxPvd *)" (?PxCreatePhysics@@YAPEAVPxPhysics@physx@@IAEAVPxFoundation@2@AEBVPxTolerancesScale@2@_NPEAVPxPvd@2@@Z)
physics_testing.exe : fatal error LNK1120: 4 unresolved externals
```

Here's my full `CMakeLists.txt` file:

```
cmake_minimum_required(VERSION 3.19)
project(physics_testing)

set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_STANDARD_REQUIRED True)

add_executable(physics_testing main.cpp)

# ---------------------------------------------

set( PHYSX_PATH ${CMAKE_CURRENT_SOURCE_DIR}/libs/PhysX/physx )

set( PHYSX_ROOT_DIR ${PHYSX_PATH} ) #This is needed for $ENV{PHYSX_PATH}/compiler/public/CMakeLists.txt
set( PHYSX_INCLUDE_DIRS {PHYSX_PATH}/include/ {PHYSX_PATH}/../pxshared/include/ )
set( PHYSX_LIBRARIES
        PhysXExtensions
        PhysX
        PhysXPvdSDK
        PhysXVehicle
        PhysXCharacterKinematic
        PhysXCooking
        PhysXCommon
        PhysXFoundation
        # SnippetUtils
        )

set(TARGET_BUILD_PLATFORM "windows") # has to match the TARGET_BUILD_PLATFORM in $ENV{PHYSX_PATH}/physix/buildtools/cmake_generate_projects.py
set(PX_BUILDSNIPPETS OFF CACHE BOOL "Generate the snippets")
set(PX_BUILDPUBLICSAMPLES OFF CACHE BOOL "Generate the samples projects")
set(PX_GENERATE_STATIC_LIBRARIES ON CACHE BOOL "Generate static libraries")
set(PX_FLOAT_POINT_PRECISE_MATH OFF CACHE BOOL "Float point precise math")
set(NV_USE_STATIC_WINCRT ON CACHE BOOL "Use the statically linked windows CRT")
set(NV_USE_DEBUG_WINCRT ON CACHE BOOL "Use the debug version of the CRT")
set(PXSHARED_PATH ${PHYSX_PATH}/../pxshared)
set(PXSHARED_INSTALL_PREFIX ${CMAKE_INSTALL_PREFIX})
set(CMAKEMODULES_VERSION "1.27")
set(CMAKEMODULES_PATH ${PHYSX_PATH}/../externals/cmakemodules)
set(PX_OUTPUT_LIB_DIR ${CMAKE_CURRENT_BINARY_DIR}/externals/physx)
set(PX_OUTPUT_BIN_DIR ${CMAKE_CURRENT_BINARY_DIR}/externals/physx)

add_subdirectory(${PHYSX_PATH}/compiler/public externals/physx)
include_directories(libs/PhysX/physx/include)
target_link_libraries(${PROJECT_NAME} ${PHYSX_LIBRARIES})
```

Here's my full `main.cpp` file:

```
#include <iostream>

#include <PxPhysicsAPI.h>

int main()
{
    physx::PxDefaultAllocator allocator{};
    physx::PxDefaultErrorCallback errorCallback{};

    const auto foundation = PxCreateFoundation(PX_PHYSICS_VERSION, allocator, errorCallback);
    const auto physics = PxCreatePhysics(PX_PHYSICS_VERSION, *foundation, physx::PxTolerancesScale(), true, nullptr);

    std::cout << "It works!" << std::endl;
    
    return 0;
}
```

What am I doing wrong? Why is the project built, but fails to link? I'm using CMake 3.19 and Visual Studio 2019 (16.11.6).

AlesBorovicka, 2022-02-07:
Hi,
 you should either include PxConfig.h as first or define a preprocessor define: PX_PHYSX_STATIC_LIB

Regards,
Ales

closed 2022-02-07 by bilek993

Pacheco95, 2022-10-19:
Thanks @AlesBorovicka. Adding `#define PX_PHYSX_STATIC_LIB` as the firs line in `main.cpp` solved the `error LNK2019: unresolved external symbol` problems.
It's curious that there is no mentions about this macro definition in the [docs](https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/search.html?q=PX_PHYSX_STATIC_LIB&check_keywords=yes&area=default).

PcChip, 2024-04-28:
Hello, 
I recently switched from Visual Studio on Windows to VSCode on Linux, and am trying to learn how to use CMake with it.  Is there a way to just type a simple include(FetchContent) section in the CMakeLists.txt and point it at the physx github page, to have it automatically download and properly include everything?  
  
For example, this works to fully include Boost and VSCode takes care of everything:  

`set(BOOST_INCLUDE_LIBRARIES math filesystem)
set(BOOST_ENABLE_CMAKE ON)
include(FetchContent)
FetchContent_Declare(
  Boost
  GIT_REPOSITORY https://github.com/boostorg/boost.git
  GIT_TAG boost-1.85.0
)`

Thanks for any suggestions you have

ayoub-belarbi, 2024-04-29:
As far as I know, currently, there is no option to do so. By the way, make sure by the way to use PhysX 5 here: https://github.com/NVIDIA-Omniverse/PhysX
PhysX 4.1 (This repo), is not actively maintained.
Thanks