NVIDIAGameWorks/PhysX #530: Is there support for rebuilding a concave/triangle mesh from previous mesh data?

kunrii, 2022-02-10:
Simple question really, if I have built a fairly large collision mesh but require changing it, is there an option to provide a previously created collision mesh to avoid a full (and lengthy) build process?

kstorey-nvidia, 2022-02-11:
See SnippetDeformableMesh

`void stepPhysics(bool /*interactive*/)
{
	{
		PxVec3* verts = gMesh->getVerticesForModification();
		gTime += 0.01f;
		updateVertices(verts, sinf(gTime)*20.0f);
		PxBounds3 newBounds = gMesh->refitBVH();
		PX_UNUSED(newBounds);

		// Reset filtering to tell the broadphase about the new mesh bounds.
		gScene->resetFiltering(*gActor);
	}
	gScene->simulate(1.0f/60.0f);
	gScene->fetchResults(true);
}`

This update function deforms the vertices of a mesh and then calls refitBVH() to do a cheap update of just the bounds of the BVH without changing the structure of the hierarchy.

Hope this helps

kunrii, 2022-02-11:
Thank you for the code snippet. Unfortunately I'm introducing new geometry so deformations alone aren't enough. I guess I'll have to find some way to chunk up the mesh without introducing a lot of overhead with expensive scene objects (this is for use with a game engine). 

Hopefully someone else will benefit from this option though, it's interesting there's a way to cheaply compute deformations that don't introduce new geometry.