NVIDIAGameWorks/PhysX #541: Should an actor be released after it is removed from the scene?

GasimGasimzada, 2022-03-03:
I am having trouble finding this information in documentation. From some experimentation, it looks like the scene does not increase the reference count of an actor:

```cpp
auto * actor = physics->createRigidStatic(transform);
scene->addActor(*actor);
actor->release(); // actor is destroyed
```

I am asking this because I can't find this information in physx visual debugger as well (all the actors are only displayed in the scene).

RobertMtx, 2022-04-17:
Based on other information I've seen in the documentation, I believe PhysX adds a reference of 1 to an element as soon as it is created. Since I'm new, I can't say for sure if being added to the scene adds a reference. If not, you may be best off using a custom list to contain all of your actors, then set up your own wrapper to add/remove scene objects. Then you can add references when they are added and subtract when they are removed.

kstorey-nvidia, 2022-04-19:
Actors are not shared resources so they are not ref counted. An actor can only be present in up to 1 scene at any given time. PxShapes, PxTriangleMesh, PxConvexMesh etc. are shared resources so are ref counted as they can be referenced by multiple actors or shapes respectively. The base type PxRefCounted is used to implement the interface for reference counting.

The semantics for actors are as follows: Actors are created via the PxPhysics class. 

You are free to add/remove actors to scenes, but it can only occupy at most one scene at any given time. 

When you call release() on an actor, it will be immediately released and additionally, if it is currently contained in a scene, it will be removed from that scene. If that scene is simulating concurrently to the release() call, then the remove/release will be deferred and processed during fetchResults(), otherwise it occurs immediately.

If you remove an actor from a scene, this does not release the actor. The actor can be inserted back into the scene or inserted into a different scene once it is no longer contained in a scene.

When a scene is released, any actors contained in the scene are automatically removed from the scene, but they are not released.

If an actor is not released before PxPhysics is released, the actor will be automatically released when PxPhysics is released. The same goes for any outstanding scenes, shapes and meshes.

Hope this helps to clarify the behavior.

GasimGasimzada, 2022-04-22:
@kstorey-nvidia That's good to know thank you!

closed 2022-04-22 by GasimGasimzada