NVIDIAGameWorks/PhysX #542: Is there a way to set a different geometry to the shape?

GasimGasimzada, 2022-03-03:
I want to be able to change the geometry of a shape (e.g from Box to Sphere). Is there a way to do it without destroying the shape and creating a new one?

RobertMtx, 2022-04-23:
I'm very new to PhysX, but I don't think you would want to change the geometry of a shape. You would create two shapes, and change the shape of a rigid body (or etc.) at runtime. So in your example, you would create box and sphere shapes, and only attach one shape to the rigid body at first. Then when you want to change shapes, attach the opposite shape, and detach the other shape.

Just make sure you keep an active reference (count) to the shapes that are not in use, to prevent them from being destroyed. You do this by not calling release() on them. Note that some of the built-in functions automatically call release after attaching a shape, so you may want to check the source code of any methods you use from the API to attach shapes to bodies. I think the method that calls release() automatically is called createExclusiveShape().

Let me know if you need any more help!

GasimGasimzada, 2022-04-27:
Yeah that's what I ended up doing. When the geometry changes, I create a new shape, detach and release the old one, and attach the new one.

closed 2022-04-27 by GasimGasimzada