NVIDIAGameWorks/PhysX #543: Unable to compile on M1 using Xcode

wulcat, 2022-03-07:
Throws error "overrides a function but not marked 'override'"

Tried almost every possible solution found. Like adding '-Wno-inconsistent-missing-override' inside Xcode c++ other flags & updating the Cmakefile.txt inside Xcode project. Still the issue seems to persist.

wulcat, 2022-03-08:
The override compiler error was solved by adding below code in physx/source/compiler/cmake/Mac/CMakeLists.txt 

--------------------------------------------------------------------------------

#TODO: Fix warnings
SET(CLANG_WARNINGS "-ferror-limit=0 -Wall -Wextra -Werror -Wno-alloca -Wno-anon-enum-enum-conversion -Wstrict-aliasing=2 -Weverything -Wno-documentation-deprecated-sync -Wno-documentation-unknown-command -Wno-gnu-anonymous-struct -Wno-undef -Wno-unused-function -Wno-nested-anon-types -Wno-float-equal -Wno-padded -Wno-weak-vtables -Wno-cast-align -Wno-conversion -Wno-missing-noreturn -Wno-missing-variable-declarations -Wno-shift-sign-overflow -Wno-covered-switch-default -Wno-exit-time-destructors -Wno-global-constructors -Wno-missing-prototypes -Wno-unreachable-code -Wno-unused-macros -Wno-unused-member-function -Wno-used-but-marked-unused -Wno-weak-template-vtables -Wno-deprecated -Wno-non-virtual-dtor -Wno-invalid-noreturn -Wno-return-type-c-linkage -Wno-reserved-id-macro -Wno-c++98-compat-pedantic -Wno-unused-local-typedef -Wno-old-style-cast -Wno-newline-eof -Wno-unused-private-field -Wno-format-nonliteral -Wno-implicit-fallthrough -Wno-undefined-reinterpret-cast -Wno-disabled-macro-expansion -Wno-zero-as-null-pointer-constant -Wno-shadow -Wno-unknown-warning-option -Wno-atomic-implicit-seq-cst -Wno-extra-semi-stmt")
SET(GCC_WARNINGS "-Wall -Werror -Wno-invalid-offsetof -Wno-uninitialized")

IF ("${CMAKE_CXX_COMPILER_ID}" STREQUAL "Clang")
	# using Clang	  
	SET(PHYSX_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -fno-rtti -fno-exceptions -ffunction-sections -fdata-sections -fstrict-aliasing ${CLANG_WARNINGS}" CACHE INTERNAL "PhysX CXX")
ELSEIF ("${CMAKE_CXX_COMPILER_ID}" STREQUAL "GNU")
	SET(PHYSX_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -fno-rtti -fno-exceptions -ffunction-sections -fdata-sections -fno-strict-aliasing ${GCC_WARNINGS}" CACHE INTERNAL "PhysX CXX")
ENDIF()

--------------------------------------------------------------------------------

Although, most of the compiler errors were gone, it's throwing errors in SnippetRenderer.h

--------------------------------------------------------------------------------

#if PX_WINDOWS
	#include <windows.h>
	#pragma warning(disable: 4505)
	#include <glut.h>
#elif PX_LINUX_FAMILY
	#include <GL/glut.h>
#elif PX_OSX
	#include <GLUT/glut.h>
#else
	#error platform not supported.
#endif

--------------------------------------------------------------------------------

while compiling on m1 Mac, the code takes us to the #error platform not supported. I commented whole part expect #include<GLUT/glut.h> but while compiling,
 it throws Undefined symbol:_glut[Most of the functions]

The-Hambone, 2022-07-31:
I am currently trying to compile on M1, did you ever figure this out? 

Also, the text to solve the first issue is weird, could you email it to me? 

I have been working on this for a couple of hours, a little frustrated.

wulcat, 2022-08-12:
Eventually I was able to compile 2 of the Xcode projects but was unable to link them all together with the core UI build. Give me sometime, will update here with the steps.