NVIDIAGameWorks/PhysX #547: What does the hammer icon displayed on the PhysX SDK website mean?

SummerInLake, 2022-03-13:
I found that both material point method (MPM) and molecular dynamics have hammer icons, which means they are still being built? Or is it planned? I didn't find anything about them in Omniverse
As far as I know, MPM is usually used in offline rendering animation, such as snow in forzen. It has a great effect, but it cannot be simulated in real time. However, I saw MPM simulated by GPU in other places, which makes real-time rendering possible. Now I see that you are also developing such a feature. I am very excited. I want to know how it is progressing?

kstorey-nvidia, 2022-03-16:
The hammer icon means "under construction". All PhysX features are still being actively iterated upon and improved, but MPM, MD and a few other features are at an earlier stage of development than the other features. The features without a hammer icon are mostly at a stage of development where they could be reliably used in an application and most of them either are - or will very soon - be available in Omniverse.

SummerInLake, 2022-03-16:
> 锤子图标的意思是“正在建设中”。所有 PhysX 功能仍在积极迭代和改进中，但 MPM、MD 和其他一些功能比其他功能处于更早的开发阶段。没有锤子图标的功能大多处于开发阶段，可以在应用程序中可靠地使用，并且其中大多数已经或将很快在 Omniverse 中可用。

> The hammer icon means "under construction". All PhysX features are still being actively iterated upon and improved, but MPM, MD and a few other features are at an earlier stage of development than [...]

Thank you for your reply. I can fully understand that your development takes time, and I will patiently wait for the arrival of these functions. However, with your efforts, I can feel that the game effect will change greatly in the future. it will be awsome!

SummerInLake, 2022-03-16:
> The hammer icon means "under construction". All PhysX features are still being actively iterated upon and improved, but MPM, MD and a few other features are at an earlier stage of development than [...]

Thank you for your reply. I can fully understand that your development takes time, and I will patiently wait for the arrival of these functions. However, with your efforts, I can feel that the game effect will change greatly in the future. it will be awsome!

SummerInLake, 2022-03-16:
> The hammer icon means "under construction". All PhysX features are still being actively iterated upon and improved, but MPM, MD and a few other features are at an earlier stage of development than [...]

Thank you for your reply. I can fully understand that your development takes time, and I will patiently wait for the arrival of these functions. However, with your efforts, I can feel that the game effect will change greatly in the future. it will be awesome!

closed 2022-03-16 by SummerInLake