NVIDIAGameWorks/PhysX #564: Geometry queries not working as expected

gabrielecaddeo, 2022-05-17:
Hi, I would like to use the API to detect compenetration between objects.
I am starting by using the `PxGeometryQuery::computePenetration` with two boxes that are compenetrating, according to their poses (`pose0` and `pose1`). 

Unfortunately, the method  `PxGeometryQuery::computePenetration` is returning 0 (i.e. no compenetration) while I was expecting the opposite.
I am attaching the code below.
```
#define NDEBUG
#include "PxPhysicsAPI.h"
#include <iostream>
using namespace physx;

PxDefaultAllocator      gAllocator;
PxDefaultErrorCallback	gErrorCallback;
PxFoundation*           gFoundation = NULL;
PxPhysics*		gPhysics = NULL;

int main (int argc, char** argv)
{
    gFoundation = PxCreateFoundation(PX_PHYSICS_VERSION, gAllocator, gErrorCallback);
    if(!gFoundation)
        std::cout<<"PxCreateFoundation failed!"<<std::endl;

    gPhysics = PxCreatePhysics(PX_PHYSICS_VERSION, *gFoundation, PxTolerancesScale(),true);
	if(!gPhysics)
		exit(0);

    PxTransform pose0 (0.0, 0.0, 0.0);
    PxTransform pose1 (0.1, 0.0, 0.0);
    PxBoxGeometry box0(0.1, 0.1, 0.1);
    PxBoxGeometry box1(0.1, 0.1, 0.1);
    PxGeometry box0_geom(box0);
    PxGeometry box1_geom(box1);

    PxVec3 direction ;
    PxF32 depth;

    bool isPenetrating = PxGeometryQuery::computePenetration(direction, depth, box0_geom, pose0, box1_geom, pose1);

    std::cout<<isPenetrating<<std::endl;
    
    gPhysics->release();
    gFoundation->release();

    return 0;
}
```

Is there something wrong that I am doing or am I missing something?

PierreTerdiman, 2022-05-17:
You don't need these two lines:

PierreTerdiman, 2022-05-17:
You don't need these two lines:

    PxGeometry box0_geom(box0);
    PxGeometry box1_geom(box1);

Just pass box0 and box1 directly to the computePenetration function instead.

gabrielecaddeo, 2022-05-17:
Thank you very much @PierreTerdiman , issue solved.

closed 2022-05-17 by gabrielecaddeo