NVIDIAGameWorks/PhysX #570: Really Hope Someone Can Answer This About PhysX Wheel Collision

ProudArts, 2022-06-08:
Hello,

I have been working on a physics based motorcycle on and off for over a year.  One of the biggest issues is the wheel collision.
If you use a sphere for the wheel it works smooth but the issue is the sphere collision sticks out the side and will hit other items or when the bike falls over the wheels will be floating in the air because of the sphere collision.

If you use a cylinder the tire works perfect at low speeds and the tire acts realist but when you start going faster the tires start to bounce off the ground see the attached videos.

Do you know how to make a cylinder in a 3D modeling program that would roll smooth?  I have tried making a few different ones in a 3D modeling program and then importing them into Unreal Engine 4.27/PhysX but so far all the ones I made bounce at high speeds.

I would really, really, really, appreciate any help you could provide, this is driving me mad :)!!!

Sincerely,
John Daniels
Proud Arts LLC

[image]
[image]

https://user-images.githubusercontent.com/44750851/172510001-f0286eae-17a0-4bb6-8afc-4838c1328823.mp4

https://user-images.githubusercontent.com/44750851/172510009-44b7669f-89e5-4fa4-aac2-b20e70171883.mp4

kstorey-nvidia, 2022-06-08:
Are you using rigid body collisions or sweeps/overlaps to generate contacts?

Are you reliant on the PhysX solver applying collision forces on the wheel or are you doing this yourself?

The issue is that, in this case, you would want the contact normal for any contact point to point exactly to the COM of the wheel so that the angular velocity does not project any velocity component onto the normal direction. As the shape is not a perfectly cylindrical shape, it is discretized and you get discrete steps on the surface of the wheel. It will roll reasonably smoothly, but as the angular velocity of the wheel increases, small offsets in the contacts can lead to significant pops due to the linear projection of the angular velocity.

There are solutions available. The first option would be to leverage contact modification - you can just take those contact points, project them onto an analytical cylindrical surface and adjust the normal to point back to the COM of the shape. If you do this, you'll get perfectly smooth contacts.

An alternative is to use PxSceneDesc::solverOffsetSlop. This is a parameter that defines a length limit for the angular projection terms used in collision (offset.cross(normal)). If the magnitude of this vector is below solverOffsetSlop, it is zeroed. In PhysX 4, this is a global setting on the scene. More recently, it's been moved to be a per-body property. It will produce a similar effect to the contact mod option above, but the contact mod option will probably produce slightly better results.

You could also look at using a slightly smaller collision geometry and setting a positive rest offset. This has the effect of simulating a sphere-swept shape. It will round edges and faces so it will reduce the amount of discretization. It won't fully solve your issues by itself, but it will make them less of an issue. The other 2 options should - in theory- solve the problems entirely.

There are other options available in PhysX 5, but they're only available through Omniverse at the moment so it probably doesn't help you much.

I hope one of these options can work for you

Kier

ProudArts, 2022-06-11:
Hello Kier,

Thank you so much for the wonderfully detailed reply, I really appreciate you!  I have been wanting to make this bike game for many years and have played with the idea on and off.  I had created my first game called NextGen SandBox for the PS4 a few years ago and was working on the sequal and then started working on this in parallel and ran into this issue and have been playing with it on and off for over a year. 

You are the only person I have found and I have asked many forums and have tried so many things but nothing has worked, I am so glad I found you Kier.  I also learned a new word, discretized,  thank you  :)   I have made so many different Convex collisions in a 3D modeling program and imported them and test and none of them have worked.  Some have definitely been better than others.

To answer your questions, the bike is made up of static meshes(rigid bodies) and physics constraints, see attached or below screenshot.   I believe that PhysX solver is the only thing applying collision forces because it is just the rigid bodies held together with Physics Constraints.  The only forces that are being applied are from angular velocity on the Physics Constraint of the rear wheel to drive forward and the handlebars to steer.  There is also impulse forces applied to the spine of the ragdoll rider to lean forward, back, left and right. These forces are manually applied with keyboard or gamepad.

Doing like you said and making the wheel collision smaller and then setting the Rest Offset to 5 did make a difference, it was not as smooth as the sphere but did allow the bike to go all the way across the map without a high speed wobble and crash.

For the contact modification, would that be done in a 3D program like Blender, Maya or 3DMax?   I would love to try this suggestion out, it sounds like it may be the perfect solution for this issue.
"You can just take those contact points, project them onto an analytical cylindrical surface and adjust the normal to point back to the COM of the shape."

Thank you again Kier for all your help,
John

PS: PhysX is a beautiful engine and is a pleasure to work with, I love working with physics simulations.

[image]
[image]

https://user-images.githubusercontent.com/44750851/173174232-c83909af-062e-4384-81a5-b5cf44e97c03.mp4

ProudArts, 2022-06-12:
Do you mean the normal on the static mesh/rigid body or on the collision for the static mesh/rigid body.  The below picture shows the normals for the wheel.
[image]

kstorey-nvidia, 2022-06-13:
I'm glad that one of my suggestions helped.

The normals I'm discussing are the contact normals that are generated by collision detection. These are chosen based on the collision features that are selected and could correspond to one of the two face normals or an edge/edge cross product direction. You can modify these values using a callback that gets invoked from PhysX just after contact generation. This is also where you can do other interesting things like reject contacts, limit forces, adjust material properties etc.

There's a discussion on implementing this in UE4 here:

https://forums.unrealengine.com/t/how-do-i-get-my-contactmodifycallback-to-work/250909

Unfortunately, there's no way to do this without writing some C++. I'm not sure to what extent UE4/UE5 supports modification - it's been something that's been supported in PhysX for as long as I can remember. It's a really powerful feature that allows you to modify the behaviour of collisions to suit your game.

Hope this helps

Kier

ProudArts, 2022-06-16:
Hi Kier,

Thank you for the additional information.  I read the link that you sent but it is a little hard for me to figure out.   Inside the Unreal Project settings there are some options.  Do you see any that could help make the wheel cylinder roll smoother?

Sincerely,
John  

[image]
[image]

[/Script/Engine.PhysicsSettings]
PhysicErrorCorrection=(PingExtrapolation=0.100000,PingLimit=100.000000,ErrorPerLinearDifference=1.000000,ErrorPerAngularDifference=1.000000,MaxRestoredStateError=1.000000,MaxLinearHardSnapDistance=400.000000,PositionLerp=0.000000,AngleLerp=0.400000,LinearVelocityCoefficient=100.000000,AngularVelocityCoefficient=10.000000,ErrorAccumulationSeconds=0.500000,ErrorAccumulationDistanceSq=15.000000,ErrorAccumulationSimilarity=100.000000)
DefaultDegreesOfFreedom=Full3D
bSuppressFaceRemapTable=False
bSupportUVFromHitResults=False
bDisableActiveActors=False
bDisableKinematicStaticPairs=False
bDisableKinematicKinematicPairs=False
bDisableCCD=False
bEnableEnhancedDeterminism=False
AnimPhysicsMinDeltaTime=0.000000
bSimulateAnimPhysicsAfterReset=False
MaxPhysicsDeltaTime=0.033333
bSubstepping=False
bSubsteppingAsync=False
MaxSubstepDeltaTime=0.016667
MaxSubsteps=6
SyncSceneSmoothingFactor=0.000000
InitialAverageFrameRate=0.016667
PhysXTreeRebuildRate=10
DefaultBroadphaseSettings=(bUseMBPOnClient=False,bUseMBPOnServer=False,bUseMBPOuterBounds=False,MBPBounds=(Min=(X=0.000000,Y=0.000000,Z=0.000000),Max=(X=0.000000,Y=0.000000,Z=0.000000),IsValid=0),MBPOuterBounds=(Min=(X=0.000000,Y=0.000000,Z=0.000000),Max=(X=0.000000,Y=0.000000,Z=0.000000),IsValid=0),MBPNumSubdivs=2)
MinDeltaVelocityForHitEvents=0.000000
ChaosSettings=(DefaultThreadingModel=TaskGraph,DedicatedThreadTickMode=VariableCappedWithTarget,DedicatedThreadBufferMode=Double)

ProudArts, 2022-06-23:
Hi Amoravanszky, thank you for the comment and the kind remarks about the motorbike.  Physics is amazing to work with.  
I have created a few threads on their forums but have not gotten any replies.  The bike tires are working a lot better with 
Kier's idea to use a slightly smaller collision geometry and setting a positive rest offset.

I never could figure out how to implantment his other 2 ideas of  "ontact modification" and "PxSceneDesc::solverOffsetSlop". 

Here is an great article that talks about this exact issue with no cylinder primitive in PhysX. 
[http://www.codercorner.com/blog/?p=1303](url)

Sincerely,
John

ProudArts, 2022-07-23:
Hey Guys,

What would be the best way to slow and stop a wheel, like a hand brake for the rear wheel?

I have played with a few different ideas:

1) Set the angular limits to locked but this goes crazy with higher speeds

2) Get the angular velocity and then apply reverse angular torque to the wheel it every frame, this works good but still not perfect

3) Set Angular damping to a very high value but it will never lock or stop the tire.

Can you think of a better way. it would be great to be able to somehow apply friction to the wheel just like a brake.

Sincerely,
John

ProudArts, 2022-07-24:
This brake is ok but does not hold the tire and is a little iractic

https://user-images.githubusercontent.com/44750851/180665707-648ac07c-7875-4a57-a747-6ad94f850106.mp4

.

closed 2022-07-27 by ProudArts