NVIDIAGameWorks/PhysX #575: Expected performance/stability difference between constrained rigidbodies and compound collider?

tysoni, 2022-06-28:
I am trying to simulate a concave ridigbody. Since PhysX does not support concave dynamic colliders, it seems the only way to simulate such a thing would be decompose it into convex parts and attach those parts together. Constraining convex parts together with completely rigid constraints (no DOF) often results in a lot of simulation instability (if a lot of contacts are generated there's a high likelihood the whole simulation will just explode). Is there a stability or performance benefit to creating a compound collider instead? Or do they work the same way internally?

kstorey-nvidia, 2022-06-28:
You can add multiple shapes to a single rigid body, which is far more efficient and stable than trying to constrain a large number of rigid bodies together with fixed joints.

Hope this helps

tysoni, 2022-06-28:
Thank you, just wanted to confirm that before rewriting my app to support it.

closed 2022-06-28 by tysoni