NVIDIAGameWorks/PhysX #593: PCMConvexVsMeshContactGeneration is using SAT?

pkusilence, 2022-08-13:
We are trying to reduce narrow phase consumption for a vehicle game with PhysX 3.4.0, and the profile shows that function "testPolyEdgeNormal" in PCMConvexVsMeshContactGeneration::generateTriangleFullContactManifold uses up to 0.2 ms. 

It seems that the function "testPolyEdgeNormal"  is calculating convex edge - triangle edge MTD normal. For a 95-face convex, in the worst case the checkedPolygonEdgeNum is 183 nbTriangleAxes 3, so for 183*3=549 times the two support function are excuted. The total calculation needs 206 us. 

[image]

[image]

AFAIK, the PCM method is different from SAT, and in the PhysX docs it is described that "PCM is a fully distance-based collision detection system". So is it really necessary to calculate the edge-edge case? If so, how could we optimize the consumption?

Pierre-Terdiman, 2022-08-15:
Hello,

As the name suggests ("_generateTriangleFullContactManifold_") this is a function that generates the full manifold. This is used even in the PCM codepath to generate the initial set of contacts - otherwise if you incrementally build these over time the object can rotate through the ground and tunnel through during that time. So yes, it is necessary to calculate the edge-edge case even with PCM.

I suppose we could look at the code and try to look for things to optimize but I'm sorry to say this is unlikely to happen for PhysX 3.4 (this version is too old at this point). Do you have the option to switch to PhysX 4.1.2 ?

pkusilence, 2022-08-15:
Hi, Pierre. Thanks for reply.

PhysX total Upgrades is not acceptable because of lack of time for game dev.  Besides, in PhysX 4.1.2 the PCM stuff is not much different from 3.4.0. 

In the testPolyEdgeNormal function , some of the edges of convex might be checked twice since the two adjacent polygon is used. So, it may help if each edge of convex is calculated once.