NVIDIAGameWorks/PhysX #597: Found bug in  Gu::sweepSphereVSTri

bleston, 2022-08-24:
I found a bug in function Gu::sweepSphereVSTri. In some case, it will miss collision.

Below code show the bug. The capsule should hit the triangle at point hitPos. But PxGeometryQuery::sweep do not find the hit.

```
void testSweepSphereTriangle()
{
	const PxVec3 vertices[3] = { {7.33294010f, 79.6109009f, -26.9337997f}, {7.61456013f, 80.2574005f, -27.3008995f }, {7.33294010f, 80.2574005f, -26.9337997f} };
	const PxU32 indices[3] = { 0, 1, 2 };

	static float radius = 0.265350759f;
	static float halfHeight = 0.340350509f;

	const PxVec3 startPoint{ 7.74110079f,80.8543167f,-27.2595482f };
	const PxQuat rot{ 0.f, 0.f,0.707106829f,0.707106829f };
	const PxVec3 unitDir{ -0.929564953f,  -0.0281119142f,  0.367584974f };
	const PxReal distance{ 0.149537891f };
	
	const PxVec3 hitPos{ 7.59219694f, 80.2574005f, -27.2717476f };
	const PxReal hitDistance{ 0.0944127962f };

	const PxVec3 p0 = hitPos - unitDir * hitDistance;

	const physx::PxTransform pose0{ startPoint, rot};
	const physx::PxTransform poseTri{ PxIdentity };

	PxHitFlags hitFlags{ physx::PxHitFlag::eDEFAULT | physx::PxHitFlag::eMESH_BOTH_SIDES | physx::PxHitFlag::eASSUME_NO_INITIAL_OVERLAP };

	PxTriangleMeshDesc meshDesc;
	meshDesc.points.count = 3;
	meshDesc.points.data = vertices;
	meshDesc.points.stride = sizeof(PxVec3);
	meshDesc.triangles.count = 1;
	meshDesc.triangles.data = indices;
	meshDesc.triangles.stride = 3 * sizeof(PxU32);
	PxCookingParams params = gCooking->getParams();

	PxTriangleMesh* triMesh = NULL;
	triMesh = gCooking->createTriangleMesh(meshDesc, gPhysics->getPhysicsInsertionCallback());

	PxTriangleMeshGeometry triGeom{ triMesh };
	PxCapsuleGeometry capsuleGeom{ radius, halfHeight };
	

	bool isInitOverlap = PxGeometryQuery::overlap(capsuleGeom, pose0, triGeom, poseTri);
	printf("isInitOverlap = %d.\n", isInitOverlap);

	physx::PxSweepHit hit;
	PxU32 hitCount = PxGeometryQuery::sweep(unitDir, distance, capsuleGeom, pose0, triGeom, poseTri, hit, hitFlags);
	printf("\nSweep capsule hitCount = %d.\n", hitCount);
	if (hitCount > 0)
	{
		printf("hit position is (%f, %f, %f).\n", hit.position.x, hit.position.y, hit.position.z);
		printf("hit normal is (%f, %f, %f).\n", hit.normal.x, hit.normal.y, hit.normal.z);
		printf("hit distance is %f.\n", hit.distance);
	}

	PxTransform pose1{ startPoint + unitDir * distance, rot };
	bool isResultOverlap = PxGeometryQuery::overlap(capsuleGeom, pose1, triGeom, poseTri);
	printf("\nisResultOverlap = %d.\n", isResultOverlap);

	PxReal p0ToCapsule = PxGeometryQuery::pointDistance(p0, capsuleGeom, pose0);
	printf("\np0(%f, %f, %f) %s capsule\n", p0.x, p0.y, p0.z, (p0ToCapsule <= 0.f) ? "is in" : "is not in");
	physx::PxRaycastHit rayHits[1];
	hitCount = PxGeometryQuery::raycast(p0, unitDir, triGeom, poseTri, distance, hitFlags, 1, rayHits);
	printf("raycast p0 hitCount = %d.\n", hitCount);
	if (hitCount > 0)
	{
		printf("raycast hit position is (%f, %f, %f).\n", rayHits[0].position.x, rayHits[0].position.y, rayHits[0].position.z);
		printf("raycast hit normal is (%f, %f, %f).\n", rayHits[0].normal.x, rayHits[0].normal.y, rayHits[0].normal.z);
[... 208 lines cut ...]
@@ -506,7 +517,7 @@ bool Gu::sweepSphereVSQuad(const PxVec3* PX_RESTRICT quadVerts, const PxVec3& no
 		if(v<0.0f)
 		{
 			// 0-2
-			TestSphere = false;
+			TestBothEdge = false;
 			e0 = 0;
 			e1 = 2;
 		}
@@ -514,11 +525,11 @@ bool Gu::sweepSphereVSQuad(const PxVec3* PX_RESTRICT quadVerts, const PxVec3& no
 		{
 			PX_ASSERT(v>=1.0f);	// Else hit quad
 			// 1-3
-			TestSphere = false;
+			TestBothEdge = false;
 			e0 = 1;
 			e1 = 3;
 		}
 	}
-	return testRayVsSphereOrCapsule(impactDistance, TestSphere, center, radius, dir, quadVerts, e0, e1);
+	return testRayVsOneOrTwoCapsule(impactDistance, TestBothEdge, center, radius, dir, quadVerts, e0, e1);
 }
 

```

Pierre-Terdiman, 2022-08-24:
Thank you, yes, I think this is a duplicate of https://github.com/NVIDIAGameWorks/PhysX/issues/501

bleston, 2022-08-25:
> Thank you, yes, I think this is a duplicate of #501

Thanks