NVIDIAGameWorks/PhysX #598: What is analog for NxActor::moveGlobalPose() in PhysX SDK 4.1?

SNMetamorph, 2022-08-30:
I'm porting old physics code from PhysX SDK 2.8.3 to 4.1 version. And I can't find modern analog for method NxActor::moveGlobalPose(). There is no information about it in migrating from 2.x to 3.x guide. Thanks!

jcarius-nv, 2022-08-31:
[`PxRigidDynamic::setKinematicTarget()`](https://github.com/NVIDIAGameWorks/PhysX/blob/93c6dd21b545605185f2febc8eeacebe49a99479/physx/include/PxRigidDynamic.h#L117) should do the job.

SNMetamorph, 2022-09-01:
Thanks! What about other functions and enums, like:
`NxActor::setLinearMomentum`
`NxActorFlags::NX_AF_DISABLE_COLLISION`

SNMetamorph, 2022-09-01:
And what about force fields which are presented in PhysX SDK 2.8?

jcarius-nv, 2022-09-05:
`PxRigidBody::setForceAndTorque` allows for different modes, e.g., applying force, impulse (momentum), and velocity change.

If you don't want a shape to partake in collisions you can set the `eSIMULATION_SHAPE`  of the shape to 0. More advanced collision setup is possible with collision filtering, there's a section about this in [the guide](https://github.com/NVIDIAGameWorks/PhysX/blob/4.1/physx/documentation/PhysXGuide/Manual/MigrationFrom28.html).

Force fields have been removed since PhysX 3.0, see release notes "[What's New In NVIDIA PhysX 3.0](https://github.com/NVIDIAGameWorks/PhysX/blob/93c6dd21b545605185f2febc8eeacebe49a99479/physx/release_notes.html#L3833)"

SNMetamorph, 2022-09-05:
@jcarius-nv thanks! Are there some replacements for force fields, or maybe it can be implemented manually?

jcarius-nv, 2022-09-06:
Unfortunately there is no replacement. It just has to be done by the user through addForce & scene queries or triggers etc.

SNMetamorph, 2022-09-27:
I'm stuck again with this old functions:
`NxShape::getGlobalPosition`
`NxShape::getGlobalOrientation`
What will be replacement in PhysX SDK 4.1 version? Please.

PierreTerdiman, 2022-09-28:
```
class PxShapeExt
{
public:
	/**
	\brief Retrieves the world space pose of the shape.

	\param[in] shape The shape for which to get the global pose.
	\param[in] actor The actor to which the shape is attached

	\return Global pose of shape.
	*/
	static PX_INLINE	PxTransform		getGlobalPose(const PxShape& shape, const PxRigidActor& actor)
	{
		return actor.getGlobalPose() * shape.getLocalPose();
	}

```

closed 2023-09-15 by SNMetamorph