NVIDIAGameWorks/PhysX #599: Building PhysX 4.1 with Visual Studio 2022 - Toolset v143 (32Bit)

RobertMtx, 2022-09-04:
I just wanted to help anyone that's trying to pull this off. I was able to create a custom preset file and modify the other build files to add this build preset. I attached the modified files. Simply copy them over the originals, then build with the new preset (vc17win32.xml).

[custom preset build.zip](https://github.com/NVIDIAGameWorks/PhysX/files/9483618/custom.preset.build.zip)

+ these files were modified from the originals to allow building with Visual Studio 2022
+ Modified files & location:

- ...\PhysX\physx\generate_projects.bat
- ...\PhysX\physx\buildtools\cmake_generate_projects.py
- ...\PhysX\physx\buildtools\presets\public\vc17win32.xml

If you would like to add 64-bit support for v143, or any other specific preset, here are the keywords to search for in the build files. Locate the keyword, then copy that related section and modify as needed.

- XboxOneToolchainVC16.txt
- DCMAKE_VS160PATH
- VS160PATH
- v142
- Install2019Dir
- VS16CL

Hope someone benefits from this!
Note - If anyone has been able to find the SDK for PhysX 5 yet, let me know how to get that!

RobertMtx, 2022-09-04:
Here are both presets (32bit & 64bit), with all default options. The other two files above (generate_projects.bat and cmake_generate_projects.py) are good as-is. Just copy those over the originals, then copy these two presets into the preset directory. This will allow you to build with toolset v143 from command line "generate_projects vc17win64" or etc.
[custom presets.zip](https://github.com/NVIDIAGameWorks/PhysX/files/9484737/custom.presets.zip)

itamar-dw, 2023-01-18:
Thanks!
I get this error during the run of `generate_projects.bat`:
```
CMake Error at C:/Users/itamar.k/PersonalFolders/Documents/code/PhysX/physx/source/compiler/cmake/CMakeLists.txt:60 (MESSAGE):
  Could not find
  C:/Users/itamar.k/PersonalFolders/Documents/code/PhysX/physx/../externals/CMakeModules
```
Any idea what goes wrong?

itamar-dw, 2023-01-18:
> **EDIT** sorry I am using the new PhysX 5.1 so I guess it won't work.. Thanks! I get this error during the run of `generate_projects.bat`:
> 
> ```
> CMake Error at C:/Users/itamar.k/PersonalFolders/Documents/code/PhysX/physx/source/compiler/cmake/CMakeLists.txt:60 (MESSAGE):
>   Could not find
>   C:/Users/itamar.k/PersonalFolders/Documents/code/PhysX/physx/../externals/CMakeModules
> ```
> 
> Any idea what goes wrong?

In case it helps anyone else, for using VisualStudio 2022 with PhysX 5.1, I used the presets given above (https://github.com/NVIDIAGameWorks/PhysX/issues/599#issuecomment-1236336851) and added the string `vc17win64` in the file `dependencies.xml` in 4 places (anywhere that `vc16win64` appears).  I didn't use the modified files from above. The created project opens and builds :)