NVIDIAGameWorks/PhysX #612: Discrepancy in SDK Guied and API reference about completion tasks

chloekek, 2022-11-06:
The [SDK Guide](https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/Manual/Simulation.html#completion-tasks) says:

> A completion task is a task that executes immediately after PxScene::simulate has exited. If PhysX has been configured to use worker threads then PxScene::simulate will start simulation tasks on the worker threads and will likely exit before the worker threads have completed the work necessary to complete the scene update. […] As a consequence, a typical completion task would first need to call PxScene::fetchResults(true) to ensure that fetchResults blocks until all worker threads started during simulate() have completed their work.

I.e. the completion task is run just before `simulate` returns. (This would also mean that completion tasks are a redundant feature, as the caller could just submit the task themselves after `simulate` returns. And since you still need a blocking `fetchResults` call, why even bother with tasks?)

* * *

The [API reference](https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxapi/files/classPxScene.html#a9a9cacecc3b0f6adaf2f3d2168c2aff5) says:

> if non-NULL, this task will have its refcount incremented in simulate(), then decremented when the scene is ready to have fetchResults called. So the task will not run until the application also calls removeReference().

I.e. the completion task is not run before all the simulation tasks have finished. (This behavior seems actually useful, as it permits asynchronous use of the API without any blocking.)

* * *

It seems to me that the SDK Guide and the API reference disagree about the behavior of completion tasks. I guess the API reference here is correct?

msauter-nvidia, 2022-11-07:
The API documentation is correct. The text in the guide is misleading and needs some clarification. Thanks for the feedback.

chloekek, 2022-11-08:
Seems to have been fixed in the documentation for PhysX 5: https://nvidia-omniverse.github.io/PhysX/physx/5.1.0/docs/Simulation.html#completion-tasks. Thanks!