NVIDIAGameWorks/PhysX #621: Heightfield modification & PVD

zemlifr, 2023-02-10:
Hello, I am using heightfield modification based on snippet provided here: https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/Manual/Geometry.html#heightfield-modification

It seem to work fine, however I don't see changes propagated to PVD which makes debugging a bit hard for me.

I can see there is some heightfield specific code in 

`void PvdMetaDataBinding::releaseAndRecreateGeometry(PvdDataStream& inStream, const PxShape& inObj, PxPhysics& /*ownerPhysics*/, PsPvd* pvd)`

which is called, but it doesn't seem to work. 
Is there something that could be fixed in physx code, or there is issue on PVD side?

pdherasciec, 2023-02-20:
Hello, heightfields get exported correctly both using PVD and OVD, would you mind sharing the code for the PVD setup? The samples are not automatically outputting to PVD nor OVD.

Here are some screens of using Kit and PVD2, to view heightfields from a heightfield UT
[image]
[image]

pdherasciec, 2023-02-20:
Realizing you want to actually modify the heightfield. Need to check up on this. I think the OVD viewer in Kit does not allow, or rather does not take such modifications into account. Need to see if to make a ticket for this.

pdherasciec, 2023-03-07:
Just FYI this is being worked on next. Thanks for exposing the issue!

MunoDevelop, 2024-07-12:
@pdherasciec hi , sorry to ask here. does pvd visualizer support real time debug render or you read after from ovd file? i tried before but nothing i can see in realtime

pdherasciec, 2024-07-12:
> @pdherasciec hi , sorry to ask here. does pvd visualizer support real time debug render or you read after from ovd file? i tried before but nothing i can see in realtime

Correct, at the moment the OVD reader in Kit/Composer that is omni.physx.pvd (or the OmniPVD extension) allows the import of an already finished OVD file.