NVIDIAGameWorks/PhysX #623: Crash on QuickHullFace::connectHalfEdges

npestov9, 2023-02-19:
There is a rare crash that happens when a convex mesh is baked at runtime. 
This is from a unity game that involves destruction, meshes are generated at runtime based on destruction and baked for convex which is when the crash happens.

```
 backtrace:
       #00  pc 0x0000000000b506b8  /data/app/lib/arm64/libunity.so (local::QuickHullFace::connectHalfEdges(local::QuickHullHalfEdge*, local::QuickHullHalfEdge*))
       #01  pc 0x0000000000b50294  /data/app/lib/arm64/libunity.so (local::QuickHullFace::mergeAdjacentFace(local::QuickHullHalfEdge*, physx::shdfnd::Array<local::QuickHullFace*, physx::shdfnd::ReflectionAllocator<local::QuickHullFace*>>&))
       #02  pc 0x0000000000b52be8  /data/app/lib/arm64/libunity.so (local::QuickHull::doAdjacentMerge(local::QuickHullFace&, bool, bool&))
       #03  pc 0x0000000000b5245c  /data/app/lib/arm64/libunity.so (local::QuickHull::addPointToHull(local::QuickHullVertex const*, local::QuickHullFace&, bool&))
       #04  pc 0x0000000000b52238  /data/app/lib/arm64/libunity.so (local::QuickHull::buildHull())
       #05  pc 0x0000000000b53ab8  /data/app/lib/arm64/libunity.so (physx::QuickHullConvexHullLib::createConvexHull())
       #06  pc 0x0000000000b3e878  /data/app/lib/arm64/libunity.so (physx::Cooking::cookConvexMeshInternal(physx::PxConvexMeshDesc const&, physx::ConvexMeshBuilder&, physx::ConvexHullLib*, physx::PxConvexMeshCookingResult::Enum*) const)
       #07  pc 0x0000000000b3eb88  /data/app/lib/arm64/libunity.so (physx::Cooking::createConvexMesh(physx::PxConvexMeshDesc const&, physx::PxPhysicsInsertionCallback&, physx::PxConvexMeshCookingResult::Enum*) const)
       #08  pc 0x00000000002b0cbc  /data/app/lib/arm64/libunity.so (CreatePxMeshFromUnityMesh(Mesh*, bool, MeshColliderCookingOptions, Matrix4x4f const&, TransformType))
       #09  pc 0x0000000000a0dcc8  /data/app/lib/arm64/libunity.so (CollisionMeshData::GetSharedNxConvexMesh(Mesh&))
       #10  pc 0x00000000002c8778  /data/app/lib/arm64/libunity.so (MeshCollider::ExtractMeshGeometry(Vector3f&, bool&))
       #11  pc 0x00000000002c89b4  /data/app/lib/arm64/libunity.so (MeshCollider::Create(Rigidbody const*))
       #12  pc 0x000000000011c468  /data/app/lib/arm64/libunity.so (MeshCollider_Set_Custom_PropSharedMesh(ScriptingBackendNativeObjectPtrOpaque*, ScriptingBackendNativeObjectPtrOpaque*))
       #13  pc 0x0000000000e76cd0  /data/app/lib/arm64/libil2cpp.so (IcoSphere_Create_m40EE7F0D2EC2B84CED8A478587275AE970698B0A)
```

preist-nvidia, 2023-02-20:
Hi @npestov9 - thank you for the report. From the stack alone this is not straightforward to debug. Any chance you can repro the issue by adapting a snippet in this repo? A cooking crash should be deterministic given the same input data.

npestov9, 2023-02-23:
Hi @preist-nvidia .
The problem is that I do not know which exact mesh could be causing this. Meshes are generated at runtime randomly. Based on the player count and the frequency that I generate these random meshes, I estimate this happens once in every 1,000,000 mesh bakings, however, this still contributes to significant amount of crashes.

The crashes happen on IOS as well and the log suggests it's on computeNormalAndCentroid
[image]

preist-nvidia, 2023-02-23:
Hi @npestov9 - could you isolate the random mesh generator into a snippet? Given the same seed, you should be able to repro the crash reliably.