NVIDIAGameWorks/PhysX #624: what is the between setDrivePosition and setDriveVelocity ?

haolly, 2023-03-02:
Doesn't setting the velocity cause it to get to a certain position?  
so why do we need another setDrivePosition API?  what is the difference between them?

preist-nvidia, 2023-03-06:
Hi @haolly - the drive will try to match both the target velocity and the target position, and push towards either target with either stiffness or damping parameters that are set. If the target velocity and position are conflicting, the drive may establish an equilibrium that is neither the target velocity nor the target position.

This might help: https://nvidia-omniverse.github.io/PhysX/physx/5.1.3/docs/Joints.html?highlight=setdriveposition#drives

closed 2023-03-06 by preist-nvidia

haolly, 2023-03-07:
@preist-nvidia  Thanks for your reply. I have one more question, the code in your links says "Drive the joint to the local(actor[0]) origin", and the document of `setDrivePosition` says "The goal is relative to the constraint frame of actor[0]", that sounds counterintuitive， does this mean the drive will only move actor[1] and remain actor[0] fixed ?  what if I have to move both actors to a specific position?

```
// Drive the joint to the local(actor[0]) origin - since no angular
// dofs are free, the angular part of the transform is ignored

d6joint->setDrivePosition(PxTransform(1.0f));
d6joint->setDriveVelocity(PxVec3(PxZero));
```

preist-nvidia, 2023-03-07:
Hi @haolly - this `PxTransform(1.0f)` is actually invalid code, we'll fix that - I think it should read `PxTransform(PxIdentity)` to set the drive to the identity transform, which is then consistent with the comment.

The drive's goal is to achieve the desired joint position, which is defined as the relative transform between the actor1 and actor0 joint frames. The drive applies forces to both actors that the joint connects, i.e. the reaction forces are properly taken into account. So if you want only actor1 to move, you can fix actor0 to the world with a fixed joint.