NVIDIAGameWorks/PhysX #631: onContactModify reports the same collision twice

pkusilence, 2023-04-12:
We found that the exactly same two shapes can generate two contact modification event  (both not in CCD).  Moreover, the two collision modification event has the mContactReportStamp, so only once the contact will be report in "onContact" callback.

[image]
[image]
[image]
[image]

pkusilence, 2023-04-20:
Last 4 days I have been debugging this problem but did not find any key clue. What I found are :
1. It only happens if the two shapes of two rigidDynamic switch the collison responese from PxFilterFlag::eSUPPRESS to  PxFilterFlag::eDEFAULT. If the two shapes are created to collide with each other at the beginning, there won't be duplicate collisions.
2. The primary scene where duplicate collisions are generated is Sc::NPhaseCore::createRbElementInteraction, I check them as the pic shown:
[image]
3. After the first duplicate collisions were found, one of the contact managers is always active, even if the AABB of the shapes are separated. When the two shapes come to be in touch again, collision will be found twice again.
4. The two duplicate collisions will go through all the pipeline of contact resolution, usually they are solved that one is given impulses to a certain value enough , and the other is given impulse nearly to zero.
5. I can filter out duplicate collisions via contact modification, and set one of them to have zero maxImpulse, but there is still a waste of narrow phase of collision detection.

pkusilence, 2023-04-20:
The problem is solved, after replacing the codes of interactions by version 3.4.2 (orginal is 3.4.0). Differences are below:
1. In constructor ShapeInteraction::ShapeInteraction, it lacks "scene.getNPhaseCore()->registerInteraction(this);" in 3.4.0;
2. In Sc::NPhaseCore::convert, as PT described, unregisterInteraction should be done before create new shapeInteraction.

closed 2023-04-20 by pkusilence

PierreTerdiman, 2023-04-20:
Glad to hear this.

Yes, 3.4.0 is quite old by now and it is likely that this is not the only bug that got fixed in more recent versions.