NVIDIAGameWorks/PhysX #632: [Question] Do not impart the based object's velocity

KristofMorva, 2023-04-13:
Hey there!

Sorry if this is a trivial question, I'm an absolute layman when it comes to PhysX (3.4): is it possible to not impart linear and angular velocity from a specific object in user-land code? So for example if I have a platform, and it stops suddenly, I'd like every object on it to remain in place and not impart the velocity of the platform they are based on (but I'd like the platform to be able to transport them).

Again, sorry for such an unprofessional question, I hope this is the right place for these.

KristofMorva, 2023-04-14:
> One option would be to set the object velocities to zero when you stop the (I assume kinematic) platform.

Thank you @preist-nvidia, that however has the undesired side effect of also stopping even if another force was applied to the object (which I'd like to keep). I have attempted to apply just enough counter-force to try to balance it, but it seems to be unstable, especially when the kinematic platform does not stop immediately, but with some transition. Knowing exactly how much linear and angular counter-force to apply seems non-trivial. I had hoped that PhysX has a nice option for it somewhere that I just can't seem to be able to find :')

preist-nvidia, 2023-04-17:
Are you controlling the kinematic slow-down? Could you not apply the corresponding per-sim-step forces to the objects on the platform then?

preist-nvidia, 2023-04-19:
What effect to these other forces that may be applied on the objects on the platform have?

KristofMorva, 2023-04-20:
> What effect do these other forces that may be applied on the objects on the platform have?

@preist-nvidia any number of forces could be applied from any direction, so i.e. there could be a force that actually pushes the object down, the object could be in constant movement (i.e. rolling in-between two forces), it might have no forces applied but still in movement (i.e. it's a sphere that's rolling due to some force that was applied before getting onto the platform), etc. In all these cases, I'd like to keep the movement of the object when the kinematic platform stops.