NVIDIAGameWorks/PhysX #637: Not getting invite email

sauravkumar8178, 2023-05-09:
I am filling the form for access Nvidia Gameworks Source but I am not getting invitation mail. 
According to eula it's taking one hours but I am not getting any response.

preist-nvidia, 2023-05-09:
What are you trying to do?

sauravkumar8178, 2023-05-09:
Okay, I am currently pursuing game development course in XAMK Open University. So I will learning more in game development and contributing my skill in open source project.

sauravkumar8178, 2023-05-09:
That why I need access

preist-nvidia, 2023-05-09:
Ok got it. Two things:
- you can get the code here without signing up, and you should consider using the new 5.1 version here: https://github.com/NVIDIA-Omniverse/PhysX
- Contributing is indirect - we look at pull requests and then merge them internally, before they will make it back to the public repo.

sauravkumar8178, 2023-05-09:
Understood, Thank you