NVIDIAGameWorks/PhysX #641: Crash on physx::shdfnd::sort

aki577, 2023-06-05:
Hi all!

Our project is currently using the latest PhysX 4.1.3 and met a weird crash on physx::shdfnd::sort in PsSort.h file.

```
PhysX_64.dll!physx::shdfnd::sort<physx::PxsContactPatch * __ptr64,physx::SortBoundsPredicateManifold,physx::shdfnd::ReflectionAllocator<physx::PxsContactPatch * __ptr64> >
PhysX_64.dll!physx::Dy::ContactReduction<6>::reduceContacts()
PhysX_64.dll!physx::Dy::PxsSolverConstraintPostProcessTask::mergeContacts
PhysX_64.dll!physx::Dy::PxsSolverConstraintPostProcessTask::runInternal()
PhysX_64.dll!physx::Cm::Task::run()
```

I have done some searches and it seems that Unity has already fixed that bug, but they didn't give any hints about that. [https://issuetracker.unity3d.com/issues/crash-in-physx-shdfnd-sort-physx-pxscontactpatch-dot-dot-dot-when-gameobject-collides-with-a-terrain-created-by-script](url)

Does anyone know about this crash and enlights me if possible?

Cheers.

preist-nvidia, 2023-06-05:
Sorry, I do not know - did you try and ask Unity? And any chance you can try to repro with PhysX 5?

aki577, 2023-06-06:
Thank you for your reply preist.
I found PhysX 5 has the same `PxSort` code in _PxSort.h_ file as PhysX 4.1.3, as well as the `reduceContacts` function in _DyDynamics.cpp_.
I am wondering the crush still happens if we repro with PhysX 5...

I will try to contact with Unity or UE in their forum.

preist-nvidia, 2023-06-06:
Hi aki,

Please let me know if you do crash with PhysX 5; and if you can provide a repro, best in an adapted snippet, we can help looking into it further.

Thank you,
Philipp