NVIDIAGameWorks/PhysX #642: Gravity force ignoring mass of dynamic bodies?

lpares12, 2023-06-13:
Hi, it seems like the gravity might be ignoring the mass of the dynamic bodies I'm creating.

First of all I initialize PhysX, create a scene, and check the gravity in it:
```
px_scene = px_physics->createScene(sceneDesc);
PxVec3 gravity = px_scene->getGravity(); // This returns (0, -9.81, 0);
```

And then I create two dynamic bodies with two shapes, calculate their mass and attach them to the scene:
```
PxGeometryHolder geometry_holder = PxBoxGeometry(PxVec3(1, 1, 1));
PxGeometryHolder geometry_holder2 = PxBoxGeometry(PxVec3(2, 2, 2));
PxMaterial px_material = px_physics->createMaterial(0.5, 0.5, 0.5);

px_shape1 = px_physics->createShape(geometry_holder.any(), &px_material, 1, false, flags);
px_shape2 = px_physics->createShape(geometry_holder.any(), &px_material, 1, false, flags);

PxReal densities[5];
densities[0] = 1;

PxReal densities2[5];
densities2[0] = 1000;

px_actor1 = px_physics->createRigidDynamic(PxTransform(PxIdentity));
px_actor1->setGlobalPose(transform) // Sets position to (0, 10, 0)
px_actor1->attachShape(*px_shape1);
PxRigidBodyExt::updateMassAndInertia(*px_actor1, &densities[0], 1);

px_actor2 = px_physics->createRigidDynamic(PxTransform(PxIdentity));
px_actor2->setGlobalPose(transform) // Sets position to (0, 10, 5)
px_actor2->attachShape(*px_shape2);
PxRigidBodyExt::updateMassAndInertia(*px_actor2, &densities2[0], 1);
px_actor2->getMass(); // Returns 1000, but I think it should be 2000 since the volume of the shape is 2x2x2m
```

And when stepping through the simulation, both bodies fall down at the same velocity, which should not be the case.

If the shape 2 has size of 2x2x2m and a density of 1000, the mass of the px_actor2 should be 2000, but the getMass call for that shape is returning 1000.

So I see two problems in my script:
1. The bodies with different masses are falling at the same velocity (both with gravity enabled).
2. UpdateMassAndInertia is not updating the mass according to the bodies.

In the docs I can't find any setting that I need to enable to make this work, is there something else I'm missing?

lpares12, 2023-06-13:
I wonder if I'm updating the shapes sizes correctly. Would it be enough to do:

```
PxGeometryHolder new_geometry = PxBoxGeometry(PxVec3(4, 4, 4));
px_shape->setGeometry(new_geometry.any());

// Recompute mass
PxRigidBodyExt::updateMassAndInertia(*px_actor1, &densities[0], 1);
```
This I'd expect to update the mass of the first actor to `512` assuming a density of 1.

preist-nvidia, 2023-06-13:
@lpares12 The issue is that your shapes are shared and not exclusive. In that case you are not allowed to update them. It is a good idea to run in debug or checked configs, which will print issues.

preist-nvidia, 2023-06-13:
I.e. you should see something like this:
```
C:\p\physx\PhysXSDK\trunk\source\physx\src\NpShape.cpp (272) : invalid parameter : PxShape::setGeometry: shared shapes attached to actors are not writable.
```

lpares12, 2023-06-13:
Thank you, this was the issue. I did see the logs, but seeing how the shape position and sizes and everything were correctly modified I still expected it to be okay.

Thanks again and sorry for opening a non-issue!

closed 2023-06-13 by lpares12