NVIDIAGameWorks/PhysX #644: Missing Binaries When Running generates_projects.sh - Unable to Link

borisbenedikter, 2023-08-03:
Hi there,

I am new to PhysX and am currently trying to set up the library on my Ubuntu 22.04 system. I cloned the repository and executed the `generate_projects.sh` script from the `physx` folder as instructed in the documentation. According to the documentation, all binaries should be located in `physx/bin/linux.clang/checked` (or debug, profile, release) after running the script. However, the only binary file I can find is `libPhysXGpu_64.so`.

My goal is to link a simple "hello world" program based on the `snippethelloworld.cpp` example to PhysX. However, when I attempt to do so (by linking only to `libPhysXGpu_64.so`), I encounter the "undefined reference" error.

I suspect that there should be other binaries in the `bin` folder to resolve this issue, but they are missing. Could someone please clarify if there should indeed be other binaries, and if so, how I can compile them or where I can find them?

Just to provide some context, I have not modified any settings in the `physx/buildtools/presets/public/linux.xml` file, so they should be at their default values.

Any assistance or guidance on resolving this issue would be greatly appreciated. Thank you!

preist-nvidia, 2023-08-03:
Hi Boris - the generate projects will only prepare the build projects (and copy the gpu shared library). You will still have to run the actual build to get the full set of binaries. Please have a look in the compiler subfolder and run make there in the corresponding config folder.

borisbenedikter, 2023-08-03:
Thank you @preist-nvidia for your prompt response and solution. Running `make` in the corresponding config folder under the compiler subfolder indeed generated the missing binaries, and I can now link my program successfully.

However, during the build process of PhysX 4 (and 5), I encountered several warnings, which caused the build to fail due to the `-Werror` flag being enabled. To resolve this, I had to remove the `-Werror` flag from the build configuration (specifically, in the file `physx/source/compiler/cmake/linux/CMakeLists.txt`). This allowed the build to complete successfully.

Once again, thanks for your help and guidance. I really appreciate it!

closed 2023-08-04 by preist-nvidia