NVIDIAGameWorks/PhysX #647: Strange read/write overlapping API calls errors

SNMetamorph, 2023-09-15:
I getting these errors spamming from PhysX (version 4.1.2), but I don't use any multithreading since my application is completely single-threaded.
[image]

Also my code of running simulation is pretty simple and looks like this
[image]
What i'm doing wrong?

msauter-nvidia, 2023-09-19:
Are you by any chance adding forces within contact callbacks? That could be a scenario that triggers this. Write operations are not allowed in contact callbacks.

SNMetamorph, 2023-09-19:
> Are you by any chance adding forces within contact callbacks? That could be a scenario that triggers this. Write operations are not allowed in contact callbacks.

@msauter-nvidia yes, exactly this happens. Interesting that I didn't found any information about write operations restrictions inside contact callbacks in documentation. 

So next, I need to do actions that now runs in contact callback. How can I do this? Create some queue, push information in it inside contact callback, and before running simulation code use this information to do actions. Will this work?

msauter-nvidia, 2023-09-20:
It is documented in PxSimulationEventCallback:

"SDK state should not be modified from within the callbacks. In particular objects should not
be created or destroyed. If state modification is needed then the changes should be stored to a buffer
and performed after the simulation step"

The solution is exactly what you described: you need to write the changes you want to do into some sort of buffer and apply them after fetchResults() has returned.

SNMetamorph, 2023-09-20:
> It is documented in PxSimulationEventCallback:
>
> "SDK state should not be modified from within the callbacks. In particular objects should not be created or destroyed. If state modification is needed [...]

Thanks

closed 2023-09-22 by SNMetamorph