NVIDIAGameWorks/PhysX #648: what is the purpose of PxCapsuleControllerDesc.material

haolly, 2023-09-17:
The documentation at https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/Manual/CharacterControllers.html#kinematic-character-controller said friction is usually either 0 or infinite, and  restitution should be avoided for CCTs

therefore, I'm confused, should I set staticFriction and dynamicFriction to 1 and restitution to 0 in this material? what is the purpose of PxCapsuleControllerDesc.material?

PierreTerdiman, 2023-09-18:
This is just the material assigned to the underlying kinematic actor. This is mostly here because the SDK expects a material for a shape, but this is indeed not used by the CCT module, as least for the behavior of characters.

haolly, 2023-09-18:
@PierreTerdiman  maybe it was used when CCT does not move? e.g. when another CCT collide with this no-moving CCT?  Because only CCT's move method take a PxControllerFilters parameter

PierreTerdiman, 2023-09-19:
CCTs are kinematic so this wouldn't be used either when CCTs collide. However I think you're right that the material parameters will be used when e.g. a dynamic rigid body collides with a non-moving CCT. In this case the material's parameters will affect the behavior of that rigid body.

closed 2023-10-06 by PierreTerdiman