NVIDIAGameWorks/PhysX #649: Error in generate_projects.bat

SigmaKlim, 2023-09-18:
When I want to generate .sln files for physx, I start generate_projects.bat as said in readme. However, the console outputs this: ""\VsWhere.exe" is not recognized as an internal or external command, operable program or batch file" at first, then runs for a bit more, then reporets a failure to locade .dll files. In the end the necessary .sln files do not appear in the proper folder. The situation repeats on my colleage computer.
Please help. Full console output provided:

```
D:\VisualStudioProjects\GameEngine\GameEngine\vendor\PhysX\physx>generate_projects.bat
Preset parameter required, available presets:
(0) vc15win64 <--- VC15 Win64 PhysX general settings
(1) vc16win64 <--- VC16 Win64 PhysX general settings
(2) vc17win64 <--- VC17 Win64 PhysX general settings
Enter preset number: 2
Running generate_projects.bat vc17win64
Running packman in preparation for cmake ...
Processing project file 'D:\VisualStudioProjects\GameEngine\GameEngine\vendor\PhysX\physx\dependencies.xml'
"\VsWhere.exe" is not recognized as an internal or external command, operable program or batch file
"\VsWhere.exe" is not recognized as an internal or external command, operable program or batch file
Using preset xml: buildtools/presets/public/vc17win64.xml
Target platform: win64 using compiler: vc17
PM_CMakeModules_PATH: D:/packman-repo/chk/CMakeModules/1.28.trunk.31965103
PM_PATHS: D:/packman-repo/chk/CMakeModules/1.28.trunk.31965103;D:/packman-repo/chk/clang-physxmetadata/4.0.0.31968624_1.1;D:/packman-repo/chk/rapidjson/1.1.0-67fac85-073453e1
Cmake: cmake.exe
Not searching for unused variables given on the command line.
-- Building for: Visual Studio 17 2022
-- Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.19045.
-- The C compiler identification is MSVC 19.37.32824.0
-- The CXX compiler identification is MSVC 19.37.32824.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.37.32822/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.37.32822/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
PHYSX ROOT D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx
PhysX Build Platform: windows
Using CXX Compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.37.32822/bin/Hostx64/x64/cl.exe
CMake Error at D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/windows/CMakeLists.txt:119 (FILE):
  FILE COPY cannot find "/bin/x86//PhysXDevice64.dll": No error.
Call Stack (most recent call first):
  D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/CMakeLists.txt:117 (INCLUDE)

CMake Error at D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/windows/CMakeLists.txt:120 (FILE):
  FILE COPY cannot find "/bin/x86//PhysXDevice64.dll": No error.
Call Stack (most recent call first):
  D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/CMakeLists.txt:117 (INCLUDE)

CMake Error at D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/windows/CMakeLists.txt:121 (FILE):
  FILE COPY cannot find "/bin/x86//PhysXDevice64.dll": No error.
Call Stack (most recent call first):
  D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/CMakeLists.txt:117 (INCLUDE)

CMake Error at D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/windows/CMakeLists.txt:122 (FILE):
  FILE COPY cannot find "/bin/x86//PhysXDevice64.dll": No error.
Call Stack (most recent call first):
  D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/CMakeLists.txt:117 (INCLUDE)

CMake Error at D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/windows/CMakeLists.txt:124 (FILE):
  FILE COPY cannot find "/bin//win64/freeglutd.dll": No error.
Call Stack (most recent call first):
  D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/CMakeLists.txt:117 (INCLUDE)

CMake Error at D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/windows/CMakeLists.txt:125 (FILE):
  FILE COPY cannot find "/bin//win64/freeglut.dll": No error.
Call Stack (most recent call first):
  D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/CMakeLists.txt:117 (INCLUDE)

CMake Error at D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/windows/CMakeLists.txt:126 (FILE):
  FILE COPY cannot find "/bin//win64/freeglut.dll": No error.
Call Stack (most recent call first):
  D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/CMakeLists.txt:117 (INCLUDE)

CMake Error at D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/windows/CMakeLists.txt:127 (FILE):
  FILE COPY cannot find "/bin//win64/freeglut.dll": No error.
Call Stack (most recent call first):
  D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/CMakeLists.txt:117 (INCLUDE)

CMake Error at D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/windows/CMakeLists.txt:148 (FILE):
  FILE COPY cannot find "/bin//win.x86_64.vc141.mt/checked/PhysXGpu_64.dll":
  No error.
Call Stack (most recent call first):
  D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/CMakeLists.txt:117 (INCLUDE)

CMake Error at D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/windows/CMakeLists.txt:149 (FILE):
  FILE COPY cannot find "/bin//win.x86_64.vc141.mt/profile/PhysXGpu_64.dll":
  No error.
Call Stack (most recent call first):
  D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/CMakeLists.txt:117 (INCLUDE)

CMake Error at D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/windows/CMakeLists.txt:150 (FILE):
  FILE COPY cannot find "/bin//win.x86_64.vc141.mt/release/PhysXGpu_64.dll":
  No error.
Call Stack (most recent call first):
  D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/CMakeLists.txt:117 (INCLUDE)

CMake Error at D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/windows/CMakeLists.txt:151 (FILE):
  FILE COPY cannot find "/bin//win.x86_64.vc141.mt/checked/PhysXGpu_64.dll":
  No error.
Call Stack (most recent call first):
  D:/VisualStudioProjects/GameEngine/GameEngine/vendor/PhysX/physx/source/compiler/cmake/CMakeLists.txt:117 (INCLUDE)

Added PhysX
Added PVDRuntime
Added Snippets
-- Configuring incomplete, errors occurred!
```

ayoub-belarbi, 2023-10-12:
Hello @SigmaKlim,
Thank you for reporting this issue.
You seem to be using PhysX 5, however you reported the issue here which is the old repo with PhysX 4. Just for tracking purposes, it will be better to report the same issue in https://github.com/NVIDIA-Omniverse/PhysX.
Second, I can't reproduce the issue from my side, but from where I'm seeing the issue seems to be in the dependencies.xml file or something related to fetching the required packages. Did you change anything there (in dependencies.xml file)?
Thank you

ayoub-belarbi, 2023-10-12:
Additionally can you try latest release in https://github.com/NVIDIA-Omniverse/PhysX?