NVIDIAGameWorks/PhysX #651: Failing asserts inside BoxManager::prepareData

SNMetamorph, 2023-09-22:
I have custom asserts handler in my project and getting these errors:
[image]
This happens not always, only on specific game levels. In addition to that, also happens some strange issues with collision too (dynamic actors collides with something in area nearby point (0, 0, 0), but there are no static/dynamic actors there).
I tried to find potential causes of this problem, and found out that problematic game level has several static actors with triangle meshes: all of actors has same global pose at point (0, 0, 0), but all geometry of these triangle meshes was presented to PhysX in world space  during cooking, not in local space as it usually happens. May this fact can be the cause of this issue?

nvtw, 2023-10-16:
Hi, I cannot say what is going wrong the information given above. Could you try to create a minimal scene that reproduces your issue? Ideally in one of the snippets that come with the PhysX SDK such that it is easily reproducible?

SNMetamorph, 2023-10-16:
I tried but could not isolate the reason of this issue. So I'll attach .pxd2 file with it to describe what actually happens. Initially box  with ID 21 lies on platform. On frame 110 platform opens and box falls into the catapult, which is a dynamic kinematic actor with ID 77. On frame 375 catapult starts moving, but for somewhat reason box falls through it on floor instead of getting movement from it. Next, box activates in-game trigger that located on floor, and it teleports box away. Exactly in this moment these asserts appears in logs (but only for first teleportation). 

There is two problems:
1. Why is box goes through catapult since it's a kinematic actor?
2. Why is that asserts are being thrown when teleportation occurs?

> In addition to that, also happens some strange issues with collision too (dynamic actors collides with something in area nearby point (0, 0, 0)

Also, I can't reproduce this anymore.

Recorded with PVD 3.2021.01
[strangebug.zip](https://github.com/NVIDIAGameWorks/PhysX/files/12918869/strangebug.zip)

SNMetamorph, 2023-11-12:
I figured out reason of these problems: game code was turning on/off eSIMULATION_SHAPE & eSCENE_QUERY_SHAPE shape flags multiple times within one frame. But interesting that with PhysX 2.8.3 this code was working well.

nvtw, 2023-11-13:
Thanks for figuring it out. Sorry that I did not write back Even after you provided the PVD 3 data, I was not able tell what's going wrong.